pauldemarco/flutter_blue #800: MTU Size Not Changed

mericgerceker, 2021-03-23:
Hi! I use 0.7.3 version. I can't change MTU size.After send "await device.connect()", I'm sending "await device.requestMtu(128)". but the automatic mtu makes the size 23.

```
D/FlutterBluePlugin( 4566): [onConnectionStateChange] status: 0 newState: 2
D/BluetoothGatt( 4566): configureMTU() - device: 50:65:83:78:FA:3D mtu: 128
D/BluetoothManager( 4566): getConnectionState()
D/BluetoothManager( 4566): getConnectedDevices
D/BluetoothGatt( 4566): discoverServices() - device: 50:65:83:78:FA:3D
D/BluetoothGatt( 4566): onSearchComplete() = Device=50:65:83:78:FA:3D Status=0
D/FlutterBluePlugin( 4566): [onServicesDiscovered] count: 4 status: 0
I/zygote64( 4566): Do full code cache collection, code=124KB, data=96KB
I/zygote64( 4566): After code cache collection, code=120KB, data=67KB
D/BluetoothGatt( 4566): onConfigureMTU() - Device=50:65:83:78:FA:3D mtu=23 status=0
D/FlutterBluePlugin( 4566): [onMtuChanged] mtu: 23 status: 0
```

toxichand, 2021-03-25:
I got the same error when change MTU then write the message, anyone knows the solution?

trungdd9, 2021-04-06:
I got the same error
> > D/BluetoothGatt(32503): configureMTU() - device: 78:A5:04:57:99:4D mtu: 512
> > D/BluetoothGatt(32503): configureMTU() - device: 78:A5:04:57:99:4D mtu: 512
> > D/BluetoothGatt(32503): onConnectionUpdated() - Device=78:A5:04:57:99:4D interval=6 latency=0 timeout=500 status=0
> > D/BluetoothGatt(32503): onConfigureMTU() - Device=78:A5:04:57:99:4D mtu=23 status=0
> >D/FlutterBluePlugin(32503): [onMtuChanged] mtu: 23 status: 0

toxichand, 2021-04-07:
@mericgerceker @trungdd9
You can fix it with few lines of code
```
device.mtu.elementAt(1).then((mtu) {
      mtu = mtu < 23 ? 20 : mtu - 3; // failsafe by always assuming an ATT MTU and not a DATA MTU
      // do your service discovery
      device.discoverServices();
    });
await device.requestMtu(512);
```
Check this PR for more detail https://github.com/pauldemarco/flutter_blue/pull/579

mericgerceker, 2021-04-08:
@toxichand At what stage should I write this code? Thank you so much :)

trungdd9, 2021-04-08:
> @mericgerceker @trungdd9
> You can fix it with few lines of code
>
> ```
> device.mtu.elementAt(1).then((mtu) {
>       mtu = mtu < 23 ? 20 : mtu - 3; // failsafe by always assuming an ATT MTU and not a DATA [...]

It doesn't work

mericgerceker, 2021-04-30:
Hello. Is there any solution? I still have problems :(

RubenKid, 2022-02-11:
Yeah, trying to increase MTU in Android device but getting 23 as maximum. Any solution? Do we know if it is something device related?. Getting this in Logs

```
D/BluetoothGatt(30607): configureMTU() - device: E0:02:97:C7:D3:DD mtu: 500
D/BluetoothGatt(30607): onConfigureMTU() - Device=E0:02:97:C7:D3:DD mtu=23 status=0
```

Thanks

cnrblm, 2022-10-28:
Same problem. Any solutions ?

tm-topnet, 2022-10-28:
We made this function.
Where d is the connected device. Sometimes request doesn't do anything. But whit this it works for us (Android)

```
Future<int> controlMtu() async {
    int mtu = await d!.mtu.first;
    await d!.requestMtu(247);

    while (mtu != 247) {
      print("Waiting for requested MTU $mtu");
      await Future.delayed(const Duration(seconds: 1));
      mtu = await d!.mtu.first;
    }
    return mtu;
  }
```

FlynnDerek, 2023-01-01:
Im facing a similar issue, on android Im able to get a larger (192) mtu for about 30 seconds before the connection(?) resets and only sends 23 bytes after. Has anyone here seen similar behavior?

[image]

FlynnDerek, 2023-01-03:
Update: setting a short delay after requesting the mtu seems to solve the above issue ^. I call the method below shortly after connecting to the device (Generic ESP32).

`Future<void> TryMtu(BluetoothDevice device) async {
  int mtuFirst = await device.mtu.first;
  print("Initial MTU is $mtuFirst bytes. Trying to negotiate 192 bytes…");
  await device.requestMtu(192);
  Future.delayed(const Duration(seconds: 2));
}`

akakiev, 2023-04-04:
> @mericgerceker @trungdd9 You can fix it with few lines of code
>
> ```
> device.mtu.elementAt(1).then((mtu) {
>       mtu = mtu < 23 ? 20 : mtu - 3; // failsafe by always assuming an ATT MTU and not a DATA [...]

Hello, can you explain me, please, what is going on at this code snippet?