pauldemarco/flutter_blue #801: Bluetooth Devices not found on one device

graemelaws-shiyani, 2021-03-25:
Hi,

I have an app that detects ble devices and connects to it. It works fine on my Sony Xperia XZ2 running android 9. However, the same build does not detect any devices on a Redmi Note 9 running android 10. Bluetooth is enabled, and the device I want to connect to shows up in the phones bluetooth settings, but flutterblue fails to return any devices in the scan results. I'm building with version 0.7.2. Any help is appreciated.

graemelaws-shiyani, 2021-03-26:
It looks like it may be caused by requiring extra location permissions on Android 10.

    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

https://stackoverflow.com/questions/58428763/android-10-not-working-with-ble-bluetooth-scanning

I will test tomorrow and update.

graemelaws-shiyani, 2021-03-30:
After uninstalling and reinstalling with added permissions, it's now working :)

Perhaps the documentation needs to be updated with these requirements?

zuboje, 2021-04-01:
@graemelaws-shiyani I tried doing the same thing, still no scan results how does your manifest file look like? I have following:

<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" /> 
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />

And Android 10 won't find anything, Android 9 is fine. Permissions are granted to access location all the time.

Ravish1, 2021-08-29:
Hi guys can you help me out .I have done the same thing above you mentioned but it failed to scan the devices even after permission added and location is ON.