pauldemarco/flutter_blue #809: Can't write more than 20 times

clem1998, 2021-04-06:
Hi,
Reading and writing was working perfectly for me and then I figured that I couldn't read more than 23 characters. In order to modify that, i used **await widget.device.requestMtu(100)**;. It works perfectly for reading but when I write it works the first 20 times (roughly) and after it doesn't work anymore. What surprises me too also is that i'm able to write one time more than 23 characters. But when I write 20 times it doesnt' work anymore. I get this error : Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null, null). I think it is due to misplacement of my device.requestMtu function but I tried different things and it didn't work.
Thanks for reading,
Clement

Here is my code if you wanna take a look.
[ble_mtu_issue.txt](https://github.com/pauldemarco/flutter_blue/files/6265523/ble_mtu_issue.txt)

mericgerceker, 2021-04-07:
I have the same problem :(

mericgerceker, 2021-05-03:
Hello. Is there any solution? I still have problems :(

clem1998, 2021-05-03:
Sorry, I still have the issue :(