pauldemarco/flutter_blue #814: device get null after assigned

muhammedkamal, 2021-04-11:
I am trying to connect to device I can find it and after assigning it's to value or add it to a list of devices I get null here is my code 
``` Future<void> getDevices() async {
/*     _devices = await FlutterBluetoothSerial.instance.getBondedDevices();
    notifyListeners(); */
    flutterBlue.stopScan();
    flutterBlue.startScan(timeout: Duration(seconds: 5));
    flutterBlue.scanResults.listen((List<ScanResult> results) {
      // do something with scan results
      for (ScanResult r in results) {
        if (!_devices.any((element) => element.id == r.device.id)) {
          _devices.insert(0, r.device);
          print('${r.device.name} found! rssi: ${r.rssi}');
          print(r.advertisementData);
          print(r.device.id);
          print(_devices.length);
          if (r.device.name == 'kamal') {
            print('found');
            targetDevice = r.device;
            print(targetDevice.name);
            flutterBlue.stopScan();
          }
        }
      }
      notifyListeners();
    });
    flutterBlue.connectedDevices.asStream().listen((event) {
      for (BluetoothDevice bd in event) {
        if (!_devices.contains(bd)) _devices.add(bd);
        print(bd.id);
      }
      notifyListeners();
    });
    print(_devices.length);
  }

  Future<bool> connect(int index) async {
    flutterBlue.stopScan();
    try {
      await targetDevice.connect();
      await discoverServices();
      // ignore: unrelated_type_equality_checks
      if (_devices[index].state == BluetoothDeviceState.connected) {
        isConnected = true;
        _connectedDevice = _devices[index];
      }
      notifyListeners();
    } catch (e) {
      print(e);
    }

    return isConnected;
  }
```
and here is the output 
4
I/BluetoothAdapter( 6327): STATE_ON
D/BluetoothLeScanner( 6327): Start Scan with callback
D/BluetoothLeScanner( 6327): onScannerRegistered() - status=0 scannerId=9 mScannerId=0
I/flutter ( 6327): kamal found! rssi: -86
I/flutter ( 6327): AdvertisementData{localName: kamal, txPowerLevel: -15, connectable: true, manufacturerData: {}, serviceData: {0000180f-0000-1000-8000-00805f9b34fb: [1]}, serviceUuids: [0000180f-0000-1000-8000-00805f9b34fb]}
I/flutter ( 6327): 7A:E8:88:94:29:5B
I/flutter ( 6327): 1
I/flutter ( 6327): found
I/flutter ( 6327): kamal
2
I/BluetoothAdapter( 6327): STATE_ON
D/BluetoothLeScanner( 6327): Stop Scan with callback
2
I/BluetoothAdapter( 6327): STATE_ON

GauravDoors4s, 2021-04-15:
I am also facing the same problem. but when I am running an example app its finding devices but in my app, it didn't.

Sergelenchimeg, 2021-11-09:
Is there anyone who solved this?