pauldemarco/flutter_blue #823: PlatformException (PlatformException(bluetooth_unavailable, bluetooth is not available, null, null))

fatinrosli, 2021-04-26:
I've tried many times, including other coding attempts and this error still comes out as if the laptop's Bluetooth can't read.
this my coding form flutterBluetoothSerial.dart
` /// State of the Bluetooth adapter.
  Future<BluetoothState> get state async => BluetoothState.fromUnderlyingValue(await _methodChannel.invokeMethod('getState'));
`

flukejones, 2021-05-01:
> FlutterBlue aims to offer the most from both platforms (iOS and Android).

The library does not support MacOS, Linux, or Windows

baconcheese113, 2022-09-23:
If you get this error with iOS/Android integration tests make sure you're not getting the instance at any point in your tests i.e
```dart
final _flutterBlue = FlutterBlue.instance;
```

instead call the instance inside of your methods directly and mock those methods for your tests