pauldemarco/flutter_blue #827: Sometimes an error message appears during transmission.

jeong77, 2021-04-29:
Sometimes an error message appears during transmission.

service.characteristics.write( data );

"PlatformException (PlatformException(write_characteristic_error, writeCharacteristic failed, null, null))"

How do I get rid of the error?

JAICHANGPARK, 2021-08-30:
I have same problem

kexuebiao, 2021-09-27:
i have the same problem too ,how to solve it

ClassicBen, 2021-11-18:
I've observed the same thing and I think I've figured out why this happens. This exception is thrown in the android when the bluetooth gat server writeCharacteristic method is called and fails to complete. I think this is due to a timing issue. If you call the flutterBlue "write" function again before the android bluetooth gat server "onCharacteristicWrite" callback is called, that's when it fails. I noticed the flutter blue write function actually returns before this "onCharacteristicWrite" is called, so it's possible to request another write before the write is actually done. We put in a 20ms timeout before doing a write request and that has removed all platform exceptions. Realistically, the flutter blue function should not return until the write is actually completed.

aminome-io, 2021-12-10:
This could be just a stupid mistake since I'm new to Flutter itself.
I always got this kind of error after the second time I connected to the device.
Canceling each stream subscription, like subscription to BluetoothDeviceState or something like that,  when I want to close a connection to the device made a fix for me.

Ignacio1110, 2022-02-25:
@ClassicBen  I think you are right.
In my situation, I did not cancel the stream subscription( for write value to the device) correctly when disconnecting the device. 
so when it auto reconnecting to the device, will occur the error.