pauldemarco/flutter_blue #829: Bluetooth Device not disconnecting

AdvaitKale01, 2021-05-01:
I am running the example application cloned from this repo. My bluetooth headphone is getting connected as expected. When I press disconnect the app says it is disconnected but the headphones are still connected to device and functional where as the app says they are disconnected.

fagnerng, 2021-05-06:
I have the same issue using a custom Bluetooth device
Device - Android Moto Z 3 Play
Android 9

Bluetooth device - ECG Strap

AdvaitKale01, 2021-06-01:
> Sounds like you are confusing two concepts 1) Pairing of your Bluetooth device with the phone/device. This is done in the OS, usually using a 6 digit number or similar. This is a long-term pairing (months). 2) Connection from the Flutter application instance to the device. This connection is made/broken when applications start, stop, device sleeps or goes out of range.
> 
> This is not a good place to get general BLE help - suggest you close the issue.

The pairing part is already done. The paired device can be connected using the package but the disconnect part is not disconnecting the device. After successful execution of the code, the device is still connected to the android device.