pauldemarco/flutter_blue #830: Unable to read all measurements on Glucose device

DivyanshuCaremonitor, 2021-05-06:
I am currently testing the "Contour Plus One" device with flutter_blue. I cannot get all the records from the device using [1,1] via sample code.

I have referred https://github.com/LiamsGitHub/GlucometerBluetoothToHealthKit and try to run the commands 

Examples of RACP commands (which are sparsely documented in the spec and that I had to discover iteratively):
[1,1] Read all records.  **NO RESPONSE**
[4,1] Number of records **GETTING RESPONCE The total size**
[1,6] Read last record received **NO RESPONSE** 
[1,5] Read first record **NO RESPONSE**
[1,3,1,45,0] Read extract from record 45 onwards **NO RESPONSE**

I also get the response when taking the live blood sample.

The steps and screenshots are as follows:-
1) Device connected
2) Expand Service 0x1808
3) Press Notify option on characteristics 0x2A18
4) Click Write option Characteristics 0x2A52 (With [1,1]) NO OUTPUT

[image]

Well, my Glucose device is having 6 records.

I am not sure what I am missing here.

Strykker007, 2021-08-23:
still having trouble reading these values?

DivyanshuCaremonitor, 2021-08-24:
Nopes, not anymore. I am able to get the values.

DivyanshuCaremonitor, 2021-08-24:
I used `characteristic.write([0x01, 0x06])` to get the latest values and changed accordingly for other values.

Strykker007, 2021-08-24:
Oh nice, I was just going to say that to solve your issue. thanks bro.

bankole175, 2021-10-31:
Hi @DivyanshuCaremonitor  @Strykker007  I'm trying to get the records recorded before connecting to an accu-chek glucometer (Instant) via flutter_blue.
Please do you have an idea of the characteristics UUID that will help me to achieve that?

Strykker007, 2021-10-31:
@bankole175, after the glucometer connected, you have run the metrod to discover services of the device, when method completes you have find service 1808 and then find 2a18 charactheristic and setNotifiy(true) and then find the 2a52 characteristic and setNotifiy(true) too, and then finally write to the 2a52 the value that you want

bankole175, 2021-10-31:
Thanks for the reply @Strykker007, I'm currently doing that but I'm only seeing the last reading I took with the glucometer

DivyanshuCaremonitor, 2021-11-01:
To get all the values from the device you have to run `characteristic.write([0x01, 0x01])`

Here you can refer to more options 👍 

[1,1] Read all records. `characteristic.write([0x01, 0x01])`
[4,1] Number of records `characteristic.write([0x04, 0x01])`
[1,6] Read last record received `characteristic.write([0x01, 0x06])`
[1,5] Read first record `characteristic.write([0x01, 0x05])`
[1,3,1,45,0] Read extract from record 45 onwards `characteristic.write([0x01, 0x03, 0x01, 0x45, 0x00])`

bankole175, 2021-11-01:
Thanks @DivyanshuCaremonitor I will give it a shot

bankole175, 2021-11-02:
@DivyanshuCaremonitor `[1,1] Read all records. characteristic.write([0x01, 0x01])` isn't returning all the records, it's only returning the last record.
Is there any other option I can explore?

Strykker007, 2021-11-08:
@bankole175 this response of glucometer is an stream, you have add a listener to your glucose characteristic to can see all results, if you don't do this, you can see only last record.

bankole175, 2022-02-14:
Hi @Strykker007 I'm just seeing this.
Please how do I add a listener to my glucose characteristics?

tiago-co, 2022-02-14:
hi @bankole175, when you write to RACP, the responde comes in sequence one by one, but this heapen to fast like a interator, when a response of a glucoleter begin reach, you must search by the characteritic 2A18 in the characteristic list,
when you found the characteristic you will receive an object BluetoothCharacteric

glucoseMeasurement.value.listen(
            (value) {
              glucoseMeasure = value[12];
              log('Evento ouvindo caracteristica');
            },
          );

this block of code still listening all measurements that the device send to app, but all measurement that reach override the previous one.

bankole175, 2022-02-17:
Thanks @tiago-co I will give it a shot