pauldemarco/flutter_blue #838: release mode error for android

cv0cv0, 2021-05-13:
I/FlutterBluePlugin(13261): setup
E/flutter (13261): [ERROR:flutter/lib/ui/ui_dart_state.cc(186)] Unhandled Exception: PlatformException(error, Field devices_ for b.d.a.m not found. Known fields are [private b.b.c.b0$i b.d.a.m.h, private static final b.d.a.m b.d.a.m.f, private static volatile b.b.c.a1 b.d.a.m.g], null, java.lang.RuntimeException: Field devices_ for b.d.a.m not found. Known fields are [private b.b.c.b0$i b.d.a.m.h, private static final b.d.a.m b.d.a.m.f, private static volatile b.b.c.a1 b.d.a.m.g]
E/flutter (13261):      at b.b.c.v0.n0(Unknown Source:72)
E/flutter (13261):      at b.b.c.v0.T(Unknown Source:698)
E/flutter (13261):      at b.b.c.v0.R(Unknown Source:12)
E/flutter (13261):      at b.b.c.k0.e(Unknown Source:60)
E/flutter (13261):      at b.b.c.k0.a(Unknown Source:49)
E/flutter (13261):      at b.b.c.d1.d(Unknown Source:17)
E/flutter (13261):      at b.b.c.d1.e(Unknown Source:4)
E/flutter (13261):      at b.b.c.z.B(Unknown Source:4)
E/flutter (13261):      at b.b.c.z$a.r(Unknown Source:9)
E/flutter (13261):      at b.b.c.z$a.q(Unknown Source:0)
E/flutter (13261):      at b.d.a.b.e(Unknown Source:1516)
E/flutter (13261):      at c.a.c.a.j$a.a(Unknown Source:17)
E/flutter (13261):      at io.flutter.embedding.engine.e.b.d(Unknown Source:57)
E/flutter (13261):      at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(Unknown Source:4)
E/flutter (13261):      at android.os.MessageQueue.nativePollOnce(Native Method)
E/flutter (13261):      at android.os.MessageQueue.next(MessageQueue.java:336)
E/flutter (13261):      at android.os.Looper.loop(Looper.java:181)
E/flutter (13261):      at android.app.ActivityThread.main(ActivityThread.java:7554)
E/flutter (13261):      at java.lang.reflect.Method.invoke(Native Method)
E/flutter (13261):      at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
E/flutter (13261):      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1020)
E/flutter (13261): )
E/flutter (13261): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:581)
E/flutter (13261): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:158)
E/flutter (13261): <asynchronous suspension>
E/flutter (13261): #2      FlutterBlue.connectedDevices.<anonymous closure> (package:flutter_blue/src/flutter_blue.dart:71)
E/flutter (13261): <asynchronous suspension>
E/flutter (13261): 
I/flutter (13261): Error starting scan.
E/flutter (13261): [ERROR:flutter/lib/ui/ui_dart_state.cc(186)] Unhandled Exception: PlatformException(startScan, Field androidScanMode_ for b.d.a.y not found. Known fields are [private int b.d.a.y.h, private b.b.c.b0$i b.d.a.y.i, private boolean b.d.a.y.j, private static final b.d.a.y b.d.a.y.f, private static volatile b.b.c.a1 b.d.a.y.g], java.lang.RuntimeException: Field androidScanMode_ for b.d.a.y not found. Known fields are [private int b.d.a.y.h, private b.b.c.b0$i b.d.a.y.i, private boolean b.d.a.y.j, private static final b.d.a.y b.d.a.y.f, private static volatile b.b.c.a1 b.d.a.y.g]
E/flutter (13261):      at b.b.c.v0.n0(Unknown Source:72)
E/flutter (13261):      at b.b.c.v0.T(Unknown Source:698)
E/flutter (13261):      at b.b.c.v0.R(Unknown Source:12)
E/flutter (13261):      at b.b.c.k0.e(Unknown Source:60)
E/flutter (13261):      at b.b.c.k0.a(Unknown Source:49)
E/flutter (13261):      at b.b.c.d1.d(Unknown Source:17)
E/flutter (13261):      at b.b.c.d1.e(Unknown Source:4)
E/flutter (13261):      at b.b.c.z$a.z(Unknown Source:9)
E/flutter (13261):      at b.b.c.z$a.y(Unknown Source:4)
E/flutter (13261):      at b.b.c.z$a.o(Unknown Source:0)
E/flutter (13261):      at b.b.c.a$a.n(Unknown Source:2)
E/flutter (13261):      at b.d.a.b.x(Unknown Source:10)
E/flutter (13261):      at b.d.a.b.e(Unknown Source:1588)
E/flutter (13261):      at c.a.c.a.j$a.a(Unknown Source:17)
E/flutter (13261):      at io.flutter.embedding.engine.e.b.d(Unknown Source:57)
E/flutter (13261):      at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(Unknown Source:4)
E/flutter (13261):      at android.os.MessageQueue.nativePollOnce(Native Method)
E/flutter (13261):      at android.os.MessageQueue.next(MessageQueue.java:336)
E/flutter (13261):      at android.os.Looper.loop(Looper.java:181)
E/flutter (13261):      at android.app.ActivityThread.main(ActivityThread.java:7554)
E/flutter (13261):      at java.lang.reflect.Method.invoke(Native Method)
E/flutter (13261):      at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
E/flutter (13261):      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1020)
E/flutter (13261): , null)
E/flutter (13261): #0      FlutterBlue.scan (package:flutter_blue/src/flutter_blue.dart:123)
E/flutter (13261): <asynchronous suspension>

BlackMonkeyidea, 2021-05-20:
Maybe you should disable the obfuscation in release mode to solve this problem.
Example For Android App Gradle Config
```Groovy
buildTypes {
        release {
            signingConfig signingConfigs.release
            shrinkResources false
            minifyEnabled false
        }
    }
```

cv0cv0, 2021-05-20:
> Maybe you should disable the obfuscation in release mode to solve this problem.
> Example For Android App Gradle Config
> 
> ```groovy
> buildTypes {
>         release {
>             signingConfig signingConfigs.release
>             shrinkResources false
>             minifyEnabled false
>         }
>     }
> ```

I add
```
-keep class com.pauldemarco.flutter_blue.** { *; }
```
worked.

closed 2021-05-20 by cv0cv0

gregk104, 2021-07-03:
BlackMoneyidea's suggestion worked.  cv0cv0, where does the '-keep' statement go?

JKornberg, 2021-11-14:
@cv0cv0 Is it possible to use the "-keep class" statement in new versions of Flutter where we have signingConfigs.release?