pauldemarco/flutter_blue #839: Can't receive notification from bluetooth device

maddionatoplus, 2021-05-14:
Good morning,
I have activated notification on a characteristic, but I can't receive any notification. 
I don't know if it's a problem of implementation of if I am missing something, but maybe you can help me.

`
await mTxCharacteristic.setNotifyValue(true);
 mTxCharacteristic.value.listen((value) {
    print("$value");
  });
`
I should receive notifications on mTxCharacteristic when I write commands to mRxCharacteristic with 

`
...
await mRxCharacteristic.write(command.toList());
...
`

 but print("$value") is never called.

Thank you in advance if you can help me.

yuriboeira11tx, 2021-05-17:
Same problem. @pauldemarco

ahoelzemann, 2021-05-18:
@maddionatoplus  check https://github.com/pauldemarco/flutter_blue/issues/841 and https://github.com/pauldemarco/flutter_blue/issues/840

maddionatoplus, 2021-05-18:
Thank you for your answer.
I already call setNotifyValue(false) before:

```
await mTxCharacteristic.setNotifyValue(false);
await desc.write([0x01, 0x00]);
await mTxCharacteristic.setNotifyValue(true);

mTxCharacteristic.value.listen((value) {
  print("$value");
});
```

and I write a descriptor to [0x01, 0x00] to set the characteristic to notification (otherwise [0x02, 0x00] is for indication). 
I didn't try to wait 0.5 in between. Isn't await enough?

Another question, in this way I should receive notification inside the listener in any moment after that time, correct?

ahoelzemann, 2021-05-18:
try this: 
```
await mTxCharacteristic.setNotifyValue(false);
await mTxCharacteristic.setNotifyValue(true);

mTxCharacteristic.value.listen((value) {
  print("$value");
});

await desc.write([0x01, 0x00]);
```

i think you need to set the listener first and afterwards call the write function.

maddionatoplus, 2021-05-19:
ok, I'm going to try. Just to be more accurate, the desc.write() is not the call that should give me the notifications, it's just to specify to the device that it should be set to notification (instead of indication).

I call the write on the characteristic (that should generate notification) after in the code (when user does a specificy interaction)

closed 2021-05-19 by maddionatoplus

maddionatoplus, 2021-06-01:
now I have problem on descriptor write with iOS:

```
*** Assertion failure in -[CBPeripheral writeValue:forDescriptor:], CBPeripheral.m:543
*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Client Characteristic Configuration descriptors must be configured using setNotifyValue:forCharacteristic:'
*** First throw call stack:
(0x1932c886c 0x1a8237c50 0x1931ce000 0x19456291c 0x1ace395a0 0x100c9572c 0x1013f2d60 0x10110407c 0x10140d370 0x1013a74d4 0x1013a9cfc 0x193244fa0 0x193244ba0 0x193243ffc 0x19323dee4 0x19323d21c 0x1aad41784 0x195c7bfe0 0x195c81854 0x100a56de4 0x192efd6b0)
libc++abi.dylib: terminating with uncaught exception of type NSException
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00000001c10b9414 libsystem_kernel.dylib`__pthread_kill + 8
libsystem_kernel.dylib`__pthread_kill:
->  0x1c10b9414 <+8>:  b.lo   0x1c10b9434               ; <+40>
    0x1c10b9418 <+12>: pacibsp 
    0x1c10b941c <+16>: stp    x29, x30, [sp, #-0x10]!
    0x1c10b9420 <+20>: mov    x29, sp
Target 0: (Runner) stopped.
```

does someone have any advice? :(