pauldemarco/flutter_blue #840: Receiving the same events twice

ahoelzemann, 2021-05-15:
Hi everyone, 

I'm having an issue, where I receive an event twice. 
I'm working with a Tracker that tracks steps and so called active minutes of the person who is wearing the device. 
Following please find the code that I execute. 

the BlowItUp() method solved an issue of getting a PlatformException after connecting and reconnecting the same device twice. The method is taken from this thread: https://github.com/pauldemarco/flutter_blue/issues/497 .

First I read the steps from my watch and I receive the data once, as it is supposed to be. But when I execute the second method to get the active minutes I always the receive the number of active minutes twice. When I swap both methods (steps and active minutes), I receive the current steps twice. 

I have absolutely  no idea what I'm doing wrong or if there is maybe a bug in flutter_blue?

thanks and best regards

```  
Future<bool> asyncInit() async {
    FlutterBlue.BlowItUp();
    flutterBlue = FlutterBlue.instance;
    await flutterBlue.stopScan();
    for (var device in await flutterBlue.connectedDevices) {
      await device.disconnect();
    }
    prefs = await SharedPreferences.getInstance();

    return true;
  }

Future<bool> getStepsAndMinutes() async {
  await Future.delayed(const Duration(seconds: 3));
  BluetoothManager bleManager = new BluetoothManager();
  await bleManager.asyncInit();
  try {
    await bleManager._connectToSavedDevice();
    await bleManager._readSteps();
    await bleManager._readMinutes();
    await bleManager.disconnectFromDevice();
  } catch (e) {
    await bleManager.disconnectFromDevice();
    return false;
  }
}

  Future<dynamic> _readMinutes() async {
    Completer completer = new Completer();
    String cmd = "";
    StreamSubscription bleSubscription;
    bool alreadyReceived = false;
    bool alreadySent = false;
    BluetoothService service = await myDevice.services.firstWhere(
        (element) => (element.uuid.toString() == ISSC_PROPRIETARY_SERVICE_UUID),
        orElse: null);
    BluetoothCharacteristic characTX = await service.characteristics.firstWhere(
        (element) => (element.uuid.toString() == UUIDSTR_ISSC_TRANS_TX),
        orElse: null);
    BluetoothCharacteristic characRX = await service.characteristics.firstWhere(
        (element) => (element.uuid.toString() == UUIDSTR_ISSC_TRANS_RX),
        orElse: null);

    print("Service found and Characteristics initialized");

    cmd = "actMins\n";
    print(Uint8List.fromList(cmd.codeUnits).toString());
    if (characTX.isNotifying) {
      await Future.delayed(const Duration(milliseconds: 500));
      await characTX.setNotifyValue(false);
    }
    if (!characTX.isNotifying) {
      await Future.delayed(const Duration(milliseconds: 500));
      await characTX.setNotifyValue(true);
    }

    Stream stream = characTX.value.timeout(Duration(seconds: 10),
        onTimeout: (timeout) async {
      await bleSubscription.cancel();
    }).asBroadcastStream(onListen: (subscription) {
      Fimber.d("Listening to Minutes Stream");
      return subscription;
    }, onCancel: (cancelEvent) {
      completer.complete(true);
      print("cancel");
    });

    bleSubscription = stream
        .listen(
          (event) async{
            if (!alreadyReceived) {
              print("Event: " + String.fromCharCodes(event));
              if (event.length > 0) {
                if (event[0] == 97 && event[4] == 105) {
                  String dd = String.fromCharCodes(event.sublist(
                      event.indexOf(61) + 1,
                      event.lastIndexOf(13))); //the number between = and \r
                  await prefs.setInt(
                      "current_active_minutes", int.parse(dd.trim()));
                  print("number of activeMinutes: " + dd.trim().toString());
                  alreadyReceived = true;
                  await bleSubscription?.cancel();
                }
              }
            }
          });
    
    if (!alreadySent) {
      print("Sending  actMins command...");
      await characRX.write(Uint8List.fromList(cmd.codeUnits),
          withoutResponse: false);
      alreadySent = true;
    }
    return await completer.future;
  }

  Future<dynamic> _readSteps() async {
    Completer completer = new Completer();
    String cmd = "";
    StreamSubscription bleSubscription;
    bool alreadyReceived = false;
    bool alreadySent = false;
    BluetoothService service = await myDevice.services.firstWhere(
            (element) => (element.uuid.toString() == ISSC_PROPRIETARY_SERVICE_UUID),
        orElse: null);
    BluetoothCharacteristic characTX = await service.characteristics.firstWhere(
            (element) => (element.uuid.toString() == UUIDSTR_ISSC_TRANS_TX),
        orElse: null);
    BluetoothCharacteristic characRX = await service.characteristics.firstWhere(
            (element) => (element.uuid.toString() == UUIDSTR_ISSC_TRANS_RX),
        orElse: null);

    print("Service found and Characteristics initialized");

    cmd = "steps\n";
    print(Uint8List.fromList(cmd.codeUnits).toString());
    if (characTX.isNotifying) {
      await Future.delayed(const Duration(milliseconds: 500));
      await characTX.setNotifyValue(false);
    }
    if (!characTX.isNotifying) {
      await Future.delayed(const Duration(milliseconds: 500));
      await characTX.setNotifyValue(true);
    }

    Stream stream = characTX.value.timeout(Duration(seconds: 10),
        onTimeout: (timeout) async {
          await bleSubscription.cancel();
        }).asBroadcastStream(onListen: (subscription) {
      Fimber.d("Listening to stream");
      return subscription;
    }, onCancel: (cancelEvent) {
      completer.complete(true);
      print("cancel");
    });

    bleSubscription = stream
        .listen(
            (event) async{
          if (!alreadyReceived) {
            print("Event: " + String.fromCharCodes(event));
            if (event.length > 0) {
              if (event[0] == 115 && event[4] == 115) {
                String dd = String.fromCharCodes(event.sublist(
                    event.indexOf(61) + 1,
                    event.lastIndexOf(13))); //the number between = and \r
                await prefs.setInt("current_steps", int.parse(dd.trim()));
                print("number of steps: " + dd.trim().toString());
                alreadyReceived = true;
                await bleSubscription?.cancel();
              }
            }
          }
        });
    if (!alreadySent) {
      print("Sending  bleSteps command...");
      await characRX.write(Uint8List.fromList(cmd.codeUnits),
          withoutResponse: false);
      alreadySent = true;
    }
    return await completer.future;
  }
```

gabrieletelesca, 2021-05-17:
I noticed the same in my app. In my case I am receiving measures and I managed to discard duplicates by using Set from dart:collection. Same happens when disconnecting device after subscribing to Bluetooth state…
Let me know if you find better solutions please

ahoelzemann, 2021-05-18:
Hi @gabrieletelesca 

I will only post the _readMinutes() function, since _readSteps() is almost similar.

I kind of solved it by changing my function as followed: 

```
Future<dynamic> _readMinutes() async {
    BluetoothCharacteristic charactx;
    Completer completer = new Completer();
    String cmd = "";
    StreamSubscription bleSubscription;
    (await myDevice.discoverServices()).forEach((service) {
      if (service.uuid.toString() == ISSC_PROPRIETARY_SERVICE_UUID) {
        service.characteristics.forEach((characteristic) {
          if (characteristic.uuid.toString() == UUIDSTR_ISSC_TRANS_TX) {
            charactx = characteristic;
          }
        });
        service.characteristics.forEach((characteristic) async {
          if (characteristic.uuid.toString() == UUIDSTR_ISSC_TRANS_RX) {
            print("Status:" + myDevice.name + " RX UUID discovered");
            print("Sending  actMins command...");

            cmd = "actMins\n";
            print(Uint8List.fromList(cmd.codeUnits).toString());
            await charactx.setNotifyValue(false);
            await Future.delayed(const Duration(milliseconds: 500));
            await charactx.setNotifyValue(true);
            bleSubscription = charactx.value
                .take(2)
                .timeout(Duration(seconds: 3), onTimeout: (timeout) async {
              bleSubscription.cancel();
              completer.complete(false);
            }).listen(
              (event) async {
                // print(event.toString() + "  //////////////");
                print("Event: " + String.fromCharCodes(event));
                if (event[0] == 97 && event[4] == 105) {
                  String dd = String.fromCharCodes(event.sublist(
                      event.indexOf(61) + 1,
                      event.lastIndexOf(13))); //the number between = and \r
                  prefs.setInt("current_active_minutes", int.parse(dd.trim()));
                  print("number of activeMinutes: " + dd.trim().toString());
                  await bleSubscription?.cancel();
                  completer.complete(true);
                }
              },
              onError: (err) {
                print('Error!: $err');
              },
              cancelOnError: true,
              onDone: () async {
                completer.complete(true);
              },
            );
            List myCmd = Uint8List.fromList(cmd.codeUnits);
            await characteristic.write(myCmd, withoutResponse: false);
          }
        });
      }
    });
    return completer.future;
  }
```

There are two differences, first I now look for the characteristics in a loop, instead of defining it as a set variable. The other thing I changed is, that I always set the notifier to false and then again true. I'm not sure if the 500 milliseconds delay is necessary, but I learned that a little delay when programming with ble is never a bad choice :).

However, I don't think that the bug is gone. But now the first incoming event contains new data instead of the old one. Still every time I reinit my BLE Class I can see in the debugger, that characTX always contains the data of the last received event. My guess is that there is a cache which is not cleared correctly.