pauldemarco/flutter_blue #845: How do you detect BLE disconnect?

thormj, 2021-05-20:
I've been doing this:
```
      print("CONNECT...");
      await widget._node.sr!.device.connect();
      widget._node.sr!.device.state.listen((event) {
        print(
            "Disconnect Detected"); //What should I do?  ToDo this stream doesnt work so well...
        setState(() {
          data['Name'] = "DISCONNECTED.";
        });
      });
    } catch (e) {
      print("EX");
      print(e);
      if (!e.toString().contains('already_connected')) {
        throw e;
      }
    } finally {
      print("READ:");
      ,,,,

```

And it isn't working the way I want:
1. When it first runs, it shows up as "disconnected." (which overwrites my name until the Read/Advise fills it in)
2. When I connect again, it seems to have a stream of things that quickly fire off in succession...?
3. [This might be a more darty/fluttery bit] -- I call this in initState after the super.initState... but it fires -before- the state gets hooked up and throws an exception.  Luckily... it seems to keep doing its thing...

So... what's the "proper" way to "Hey if my BLE device is disconnected, throw up an alert and then go back to the main page"?

nmbus, 2021-05-30:
Well what happens is you are somehow not looking at the "event" that is passed in your listener, which is the state of the device. The list of possible states is listed in the documentation https://pub.dev/documentation/flutter_blue/latest/flutter_blue/BluetoothDeviceState-class.html

thormj, 2021-06-11:
Ah... that was it.  I needed to RTFM a bit more.... thanks!

closed 2021-06-11 by thormj