pauldemarco/flutter_blue #851: java.lang.NullPointerException: Attempt to invoke virtual method 'int android.util.SparseArray.size()'  error

MohamadMerei1, 2021-05-26:
Everything is fine with my bluetooth app, except for the following error that keeps causing my app to crash.

E/AndroidRuntime(26544): FATAL EXCEPTION: main
E/AndroidRuntime(26544): Process: com.example.flutter_app, PID: 26544
E/AndroidRuntime(26544): java.lang.NullPointerException: Attempt to invoke virtual method 'int android.util.SparseArray.size()' on a null object reference
E/AndroidRuntime(26544): 	at com.pauldemarco.flutter_blue.ProtoMaker.from(ProtoMaker.java:71)
E/AndroidRuntime(26544): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin$2.onScanResult(FlutterBluePlugin.java:779)
E/AndroidRuntime(26544): 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:616)
E/AndroidRuntime(26544): 	at android.os.Handler.handleCallback(Handler.java:938)
E/AndroidRuntime(26544): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/AndroidRuntime(26544): 	at android.os.Looper.loop(Looper.java:246)
E/AndroidRuntime(26544): 	at android.app.ActivityThread.main(ActivityThread.java:8512)
E/AndroidRuntime(26544): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(26544): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
E/AndroidRuntime(26544): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
E/ScanRecord(26544): unable to parse scan record: [2, 1, 6, 26, -1, 76, 0, 2, 21, -30, -59, 109, -75, -33, -5, 72, -46, -80, 96, -48, -11, -89, 16, -106, -32, 0, 0, 0, 0, -59, 7, 9, 66, 108, 117, 110, 111, 50, 5, 18, 8, 0, 8, 0, 5, 2, -80, -33, 18, 24, 2, 10, 0, 2, -1, 110, 0, 0, 0, 0, 0, 0]

JonYun, 2021-05-28:
I also encountered the same situation as you. Did you solve this problem?

JonYun, 2021-05-28:
I am being troubled by this problem and hope to get help

MohamadMerei1, 2021-05-28:
> I also encountered the same situation as you. Did you solve this problem?

No, I have not got it solved, but for me it happens randomly, my app would be working fine for few days and sometimes hours, and suddenly crashes and i keep recieving this error, i never seen this error untill i updated to a new version of flutter_blue, when i get this error, i try building the code again, uninstalling the app and restarting my phone, those steps sometimes help in getting my app back to normal, but still at anytime my app would crash.

jahnli, 2021-09-13:
@MohamadMerei1 @JonYun   Whether the problem has been solved

MohamadMerei1, 2021-09-13:
> @MohamadMerei1 @JonYun Whether the problem has been solved

Hi, i do not think i got that to work from what i rmb, plus i have not really been working on this for a while, if this happens to you randomly and not evertime, i think you can try re-installing the app once it happens, flutter bluetooth is problamatic, and i would suggest you use react native as your app development software.

JonYun, 2021-09-13:
The error I encountered was caused by the unqualified broadcast of bluetooth device. Finally, I solved the error with the help of hardware engineers

jahnli, 2021-09-13:
@JonYun  但是这个问题在扫描蓝牙设备时就出现了错误，I don't feel that connected to the device

jahnli, 2021-09-13:
If not, should also prompt error, rather than directly application crash, do you have a way to intercept the error   @JonYun

darshansoni09, 2024-08-13:
I'm facing the same issue. Could you let me know if you have any solution?