pauldemarco/flutter_blue #852: Platform Exception Field state_ for b.i.a.o not found

rkunboxed, 2021-05-27:
I got this strange error (reported in Sentry) when calling

```
flutter_blue.dart in FlutterBlue.state.<fn> at line 59
Stream<BluetoothState> get state async* {...}
```

Here is the error
```
PlatformException(error, Field state_ for b.i.a.o not found. Known fields are [private static final b.i.a.o b.i.a.o.f, private static volatile b.g.d.a1 b.i.a.o.g], null, java.lang.RuntimeException: Field state_ for b.i.a.o not found. Known fields are [private static final b.i.a.o b.i.a.o.f, private static volatile b.g.d.a1 b.i.a.o.g]
	at b.g.d.v0.a(Unknown Source:72)
	at b.g.d.v0.a(Unknown Source:715)
	at b.g.d.v0.a(Unknown Source:12)
	at b.g.d.k0.a(Unknown Source:60)
	at b.g.d.k0.a(Unknown Source:49)
	at b.g.d.d1.a(Unknown Source:17)
	at b.g.d.d1.a(Unknown Source:4)
	at b.g.d.z.m(Unknown Source:4)
	at b.g.d.z$a.e(Unknown Source:9)
	at b.g.d.z$a.d(Unknown Source:0)
	at b.i.a.b.onMethodCall(Unknown Source:1482)
	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(Unknown Source:17)
	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(Unknown Source:57)
	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(Unknown Source:4)
	at android.os.MessageQueue.nativePollOnce(Native Method)
	at android.os.MessageQueue.next(MessageQueue.java:336)
	at android.os.Looper.loop(Looper.java:197)
	at android.app.ActivityThread.main(ActivityThread.java:8107)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:496)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1100)
```
	
I've only seen it once but any idea what could be causing this? I'm running 0.7.3.

phamconganh, 2021-06-23:
in your 'proguard-rules.pro' file
```
-keep class com.pauldemarco.flutter_blue.** { *; }
```

Rexios80, 2021-08-19:
This needs to be in the readme

rkunboxed, 2022-02-08:
@phamconganh thank you so much, that fixed things right up for me!

closed 2022-02-08 by rkunboxed