pauldemarco/flutter_blue #854: Direct connection via BT id

jay2jp, 2021-05-28:
I read through the documentation and I could not see a way to directly connect to devices if my app knows BT MAC address and the connections credentials?  I am researching if I could use this plugin to implement an NFC pairing function via this library! If it's not available I would be willing to help create it!  If I missed it while going through the documentation where can I find this info?

nmbus, 2021-05-30:
Do you intend to use this only on Android or iOS as well ? iOS does not have the ability (or rather does not let you) get the MAC address of the remote bluetooth device.
Then you could always just start a scan and filter for the MAC address that was exchange via your NFC pairing.

jay2jp, 2021-06-02:
that's actually what I ended up doing due to that exact reason thanks @nmbus

closed 2021-06-02 by jay2jp