pauldemarco/flutter_blue #855: Unhandled Exception: PlatformException(error, Field devices_ for c.d.a.m not found. Known fields are

yuriboeira11tx, 2021-05-31:
This error happens when I build the **release version** of my app.

More:

```
E/flutter (30791): [ERROR:flutter/lib/ui/ui_dart_state.cc(199)] Unhandled Exception: PlatformException(error, Field devices_ for c.d.a.m not found. Known fields are [private c.b.c.b0$i c.d.a.m.h, private static final c.d.a.m c.d.a.m.i, private static volatile c.b.c.a1 c.d.a.m.j], null, java.lang.RuntimeException: Field devices_ for c.d.a.m not found. Known fields are [private c.b.c.b0$i c.d.a.m.h, private static final c.d.a.m c.d.a.m.i, private static volatile c.b.c.a1 c.d.a.m.j]
E/flutter (30791): 	at c.b.c.v0.n0(Unknown Source:72)
E/flutter (30791): 	at c.b.c.v0.T(Unknown Source:697)
E/flutter (30791): 	at c.b.c.v0.R(Unknown Source:12)
E/flutter (30791): 	at c.b.c.k0.e(Unknown Source:60)
E/flutter (30791): 	at c.b.c.k0.a(Unknown Source:49)
E/flutter (30791): 	at c.b.c.d1.d(Unknown Source:17)
E/flutter (30791): 	at c.b.c.d1.e(Unknown Source:4)
E/flutter (30791): 	at c.b.c.z.B(Unknown Source:4)
E/flutter (30791): 	at c.b.c.z$a.r(Unknown Source:9)
E/flutter (30791): 	at c.b.c.z$a.q(Unknown Source:0)
E/flutter (30791): 	at c.d.a.b.g(Unknown Source:1511)
E/flutter (30791): 	at d.a.c.a.j$a.a(Unknown Source:17)
E/flutter (30791): 	at io.flutter.embedding.engine.e.b.d(Unknown Source:57)
E/flutter (30791): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(Unknown Source:4)
E/flutter (30791): 	at android.os.MessageQueue.nativePollOnce(Native Method)
E/flutter (30791): 	at android.os.MessageQueue.next(MessageQueue.java:326)
E/flutter (30791): 	at android.os.Looper.loop(Looper.java:160)
E/flutter (30791): 	at android.app.ActivityThread.main(ActivityThread.java:6912)
E/flutter (30791): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter (30791): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
E/flutter (30791): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:860)
E/flutter (30791): )
E/flutter (30791): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:597)
E/flutter (30791): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:158)
E/flutter (30791): <asynchronous suspension>
E/flutter (30791): #2      _HomeStoreBase.getDevices (package:jiga_app/stores/home_store.dart:46)
E/flutter (30791): <asynchronous suspension>
E/flutter (30791): #3      AsyncAction.run (package:mobx/src/api/async/async_action.dart:28)
E/flutter (30791): <asynchronous suspension>
```

yuriboeira11tx, 2021-05-31:
I solved this temporarily by defining it in the **build.gradle**

```
buildTypes {
        release {
            **minifyEnabled false
            shrinkResources false**
            // TODO: Add your own signing config for the release build.
            // Signing with the debug keys for now, so `flutter run --release` works.
            signingConfig signingConfigs.debug
        }
 }
```

heroqin, 2021-06-10:
I got the same problem in my app release version. I did temporarily solve the problem by  modifying the build.gradle, but the size of apk grows.

yuriboeira11tx, 2021-06-17:
Idem!

mrdishant, 2021-07-20:
Any update on this?

yuriboeira11tx, 2021-07-20:
Not.

shuizhuyu1015, 2021-08-08:
> 我通过在**build.gradle 中**定义它暂时解决了这个问题****
> 
> ```
> buildTypes {
>         release {
>             minifyEnabled false
>             shrinkResources false
>             // TODO: Add your own signing config for the release build.
>             // Signing with the debug keys for now, so `flutter run --release` works.
>             signingConfig signingConfigs.debug
>         }
>  }
> ```

It took me a long time, but resolved my issue here!

Cyril212, 2021-08-12:
You could solve it by excluding this lib from obfuscation. 

Just add this line to proguard-project.txt

-keep class com.pauldemarco.flutter_blue.* { *; }