pauldemarco/flutter_blue #859: Sending more than 20 bytes of data

mericgerceker, 2021-06-03:
Hi! We still can't send more than 20 bytes of data with the library. Is there a solution to this? We really need it.

AlainHsu, 2021-06-08:
Hi, you can try to extend the mtu, check this https://github.com/iamobj/auto-vacc

nmbus, 2021-06-10:
As @AlainHsu said, you can use requestMtu to increase the MTU on Android devices (Up to 512 bytes). On iOS, it'll depend on whether or not both the Data Length Extension (DLE), which sets up the MTU at 512 bytes as well (185 if not). If you're gonna transmit arbitrarely large amounts of data though, I'd suggest building a layer on top that uses characteristics as a transport layer.

mericgerceker, 2021-06-11:
@AlainHsu @nmbus Hi! When I run the requestMtu method, I get this result. :(

`D/BluetoothGatt(32503): configureMTU() - device: 78:A5:04:57:99:4D mtu: 512
D/BluetoothGatt(32503): configureMTU() - device: 78:A5:04:57:99:4D mtu: 512
D/BluetoothGatt(32503): onConnectionUpdated() - Device=78:A5:04:57:99:4D interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(32503): onConfigureMTU() - Device=78:A5:04:57:99:4D mtu=23 status=0
D/FlutterBluePlugin(32503): [onMtuChanged] mtu: 23 status: 0`

AlainHsu, 2021-06-11:
@mehmetf unfortunately, it looks like your device does not support to extend the MTU. You may need to check with the hardware developers of the device.

Here is some reference: https://stackoverflow.com/a/65897345/6233187

ertandagdelen, 2021-07-06:
It worked for me. Add to connect device function.

` await targetDevice?.connect(autoConnect: true);
    if (!Platform.isIOS) {
      targetDevice!.mtu.elementAt(1).then((mtu) {
        mtu = mtu < 23
            ? 20
            : mtu -
                3; // failsafe by always assuming an ATT MTU and not a DATA MTU
        // do your service discovery
        targetDevice!.discoverServices();
      });
      await targetDevice!.requestMtu(185);
    }`