pauldemarco/flutter_blue #862: MissingPluginException: while using background refresh with flutter_background_service

Sheetal-25, 2021-06-04:
It's working on a button click but when I try to call it in the background it causes an error. Please help me out. tried from this error

Pubspec.yaml

flutter_background_service: ^0.1.5
flutter_blue: ^0.8.0

Unhandled Exception: MissingPluginException(No implementation found for method isAvailable on channel plugins.pauldemarco.com/flutter_blue/methods)

[image]

konopka90, 2021-06-14:
Same here.
I'm trying to integrate flutter_blue with firebase_messaging where I want to start scanning for BLE devices.
Seems that flutter_blue is not working with top-level functions.

[Firebase messaging top-level background function call](https://firebase.flutter.dev/docs/messaging/usage/#background-messages)

Sheetal-25, 2021-06-18:
Hey @konopka90 
I used flutter_reactive_ble: ^3.1.1+1 instead of flutter_blue:^0.8.0. it's work for me.

Have you found any other solutions?

konopka90, 2021-06-21:
@Sheetal-25 Thank you for advice. Finally I used reactive ble and didn't experiment more with flutter_blue.

adailydose, 2021-06-23:
Sounds like no luck with flutter_blue and background processes then? But background ble works with flutter_reactive_ble?

keenhon, 2021-07-22:
@konopka90 @Sheetal-25 Which package did you use to run flutter_reactive_ble in background mode? Any sample code?