pauldemarco/flutter_blue #864: Support new Android 12 Bluetooth permissions

navaronbracke, 2021-06-07:
Android 12 introduces new Bluetooth permissions:
`BLUETOOTH_SCAN` & `BLUETOOTH_CONNECT`
https://developer.android.com/about/versions/12/features/bluetooth-permissions

The plugin should update its `manifest.xml` to match the new format outlined in the documentation.
The documentation in the README should reflect this change.

It should also update its permission handling.
On Android 11 and lower: Use the current permission request logic.
On Android 12+: Use permission logic that matches the required behavior:

When calling `startScan()` we should request permission for `BLUETOOTH_SCAN`, prior to actually starting the scan.
When calling `device.connect()` we should request permission for `BLUETOOTH_CONNECT`.

dolan23, 2021-07-15:
Strong agree!

espresso3389, 2021-09-19:
My own flutter_blue fork that supports Android 12 permission:
https://github.com/espresso3389/flutter_blue

It's still under development but the example code seems work correctly.
I'm preparing PR but the changes are relatively large and I'm not sure it can be safely merged or not...

MrCsabaToth, 2021-12-24:
Could you fork off of your repo from this https://github.com/boskokg/flutter_blue/ repo or make a PR against it? That repo has many more PRs https://github.com/pauldemarco/flutter_blue/issues/929#issuecomment-930759129

MrCsabaToth, 2021-12-24:
Android 12 support is badly needed.

MrCsabaToth, 2021-12-24:
Sorry, look like the fork I mentioned already has 12 support.

tricaricom, 2022-01-10:
How do I use the espresso3389 fork in my Flutter project instead of the pauldemarco one?

espresso3389, 2022-01-10:
@tricaricom
Do like the following on your pubspec.yaml:

```yaml
  flutter_blue:
    git: https://github.com/espresso3389/flutter_blue
```

tricaricom, 2022-01-10:
Thanks @espresso3389 .

So when I use your flutter_blue, I get this error:

E/AndroidRuntime(15548): java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for android.content.AttributionSource@41052d27: AdapterService getRemoteName

However, I have these permissions in the manifest:

    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>

espresso3389, 2022-01-10:
@tricaricom 
Same issue is reported [here](https://github.com/pauldemarco/flutter_blue/pull/940#issuecomment-990451003) but I could not reproduce the issue with flutter_blue sample and my own codes...

Can you provide me with the complete sample that causes the issue?

ekuleshov, 2022-01-10:
@espresso3389 what repo/branch should I use for testing?

espresso3389, 2022-01-10:
@ekuleshov I'm working on "my" master branch, so the following lines are enough:

```yaml
  flutter_blue:
    git: https://github.com/espresso3389/flutter_blue
```

espresso3389, 2022-01-10:
@ekuleshov The plugin (on the fork) basically requires both `compileSdkVersion` and `targetSdkVersion` set to 31:

https://github.com/espresso3389/flutter_blue/blob/80fca568e59428722490a8cae3c56a039c4dc283/android/build.gradle#L31

And, the example also does the same.

I'm not sure what happens if you set `targetSdkVersion 30` on your code...

ekuleshov, 2022-01-10:
> I'm not sure what happens if you set `targetSdkVersion 30` on your code...

I already tried to bump the targetSdkVersion to 31 in my app, but still getting the same error for CONNECT permission.

If I go to the system settings, remove nearby permission and then grant it again the scan in my app is working after that. So, I still think both permissions need to be requested for the scan. Note the annotation on the `BluetoothDevice.getName()` which is called by scan.

espresso3389, 2022-01-10:
@ekuleshov

> I still think both permissions need to be requested for the scan. Note the annotation on the BluetoothDevice.getName() which is called by scan.

Ah, I see your point. I'll update the permission logic to deal with the permissions.

MrCsabaToth, 2022-01-10:
It'd be so great if https://github.com/espresso3389/flutter_blue/ would contribute to https://github.com/boskokg/flutter_blue/ to unify efforts. boskokg also has null safety and permissions but a whole bunch more PRs.

espresso3389, 2022-01-10:
@MrCsabaToth Yes, it's good if he can do so easily. There seems at least two problems:
1. He seems so busy
2. SDK 31 breaks so many things; some of them need changes on other Flutter plugins

espresso3389, 2022-01-10:
@ekuleshov Try the latest commit ([f0d9cb](https://github.com/espresso3389/flutter_blue/commit/f0d9cb88a36c48ab6f3d49861f76cec6132b2a6f)), I've changed the way to request/check permissions on scan. (I actually could not reproduce the issue but I think it handles the situations better).

ekuleshov, 2022-01-10:
> @ekuleshov Try the latest commit ([f0d9cb](https://github.com/espresso3389/flutter_blue/commit/f0d9cb88a36c48ab6f3d49861f76cec6132b2a6f)), I've changed the way to request/check permissions on scan. (I actually could not reproduce the issue but I think it handles the situations better).

Somehow it got worse. No permission request on a fresh install and I get this error now when scan() is called:

```
E/flutter ( 5232): [ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: PlatformException(startScan, Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@7c328bd: GattService registerScanner, java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@7c328bd: GattService registerScanner
E/flutter ( 5232): 	at android.os.Parcel.createExceptionOrNull(Parcel.java:2425)
E/flutter ( 5232): 	at android.os.Parcel.createException(Parcel.java:2409)
E/flutter ( 5232): 	at android.os.Parcel.readException(Parcel.java:2392)
E/flutter ( 5232): 	at android.os.Parcel.readException(Parcel.java:2334)
E/flutter ( 5232): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.registerScanner(IBluetoothGatt.java:1727)
E/flutter ( 5232): 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.startRegistration(BluetoothLeScanner.java:426)
E/flutter ( 5232): 	at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:278)
E/flutter ( 5232): 	at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:154)
E/flutter ( 5232): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin.startScan21(FlutterBluePlugin.java:846)
E/flutter ( 5232): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin.startScan(FlutterBluePlugin.java:782)
E/flutter ( 5232): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin.lambda$onMethodCall$0$FlutterBluePlugin(FlutterBluePlugin.java:246)
E/flutter ( 5232): 	at com.pauldemarco.flutter_blue.-$$Lambda$FlutterBluePlugin$npO5ZnMb5Xa1WrWJ1kqQqzqyO7E.op(Unknown Source:6)
E/flutter ( 5232): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin.ensurePermissionsBeforeAction(FlutterBluePlugin.java:666)
E/flutter ( 5232): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin.onMethodCall(FlutterBluePlugin.java:244)
E/flutter ( 5232): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:262)
E/flutter ( 5232): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:178)
E/flutter ( 5232): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$handleMessageFromDart$0$DartMessenger(DartMessenger.java:206)
E/flutter ( 5232): 	at io.flutter.embedding.engine.dart.-$$Lambda$DartMessenger$6ZD1MYkhaLxyPjtoFDxe45u43DI.run(Unknown Source:12)
```

espresso3389, 2022-01-10:
@ekuleshov Sorry, the condition is completely reversed... Please try [256f1a](https://github.com/pauldemarco/flutter_blue/pull/940/commits/256f1aad3cc7e0d8ef6d4af8da713a8efd402699).

ekuleshov, 2022-01-10:
> @ekuleshov Sorry, the condition is completely reversed... Please try [256f1a](https://github.com/pauldemarco/flutter_blue/pull/940/commits/256f1aad3cc7e0d8ef6d4af8da713a8efd402699).

Success! Thank you.

tricaricom, 2022-01-17:
The @espresso3389 fork does not work properly on BOTH pre-Android 12 and post-Android 12 devices.   I can get this to work fine on my Android 12 device.  But when I test it on Android 9, it does discover the bluetooth device, but when I try to connect, I get this error:

E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(18144): Failed to handle method call
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(18144): java.lang.NullPointerException: Attempt to get length of null array

ekuleshov, 2022-01-18:
It seems like it needs some better `null` handling. It passes `null` to `allPermissionsGranted()` which returns `false` and then it calls `ActivityCompat.requestPermissions()` on the same array. Either it should not use `null` at the line 654 (return an empty `String[]` instead) or either return `true` from the `allPermissionsGranted()` when argument is `null`.

espresso3389, 2022-01-18:
@ekuleshov Now `allPermissionsGranted` returns `null` for `permissions==null` case.

ekuleshov, 2022-01-18:
> @ekuleshov Now `allPermissionsGranted` returns `null` for `permissions==null` case.

That works, thank you! Though from the error-proneness point of view it would be good to avoid using nulls values for anything. But that is also somewhat matter of a code style.

MrCsabaToth, 2022-01-24:
> @MrCsabaToth Yes, it's good if he can do so easily. There seems at least two problems:
> 
>     1. He seems so busy
> 
>     2. SDK 31 breaks so many things; some of them need changes on other Flutter plugins
> 
> 
> Anyway, PR #940 is for the issue.

What I thought is it'd be so awesome if your fork was forked from `boskokg` (https://github.com/boskokg/flutter_blue/ - which is a fork of pauldemarco) instead of directly from pauldemarco, because `boskokg` has a lot of PR merged.

boskokg, 2022-01-24:
@espresso3389 thank you. tomorrow I will check and merge it and increase the version.

maddionatoplus, 2022-01-31:
if could be helpful, I've added a permission check before scan:

`
    var scanStatus = await Permission.bluetoothScan.request().isGranted;
    var bluetoothConnectStatus =
        await Permission.bluetoothConnect.request().isGranted;

    if (scanStatus && bluetoothConnectStatus) {
      flutterBlue.startScan(
          timeout: const Duration(seconds: Constants.bluetoothScanPeriod));

`
using this library: 

https://pub.dev/packages/permission_handler

maddionatoplus, 2022-02-01:
Yes, I didn't think about it. I didn't find another solution, the @espresso3389 one didn't work for me

ekuleshov, 2022-02-01:
> Yes, I didn't think about it. I didn't find another solution, the @espresso3389 one didn't work for me

He made a few fixes and his latest changes seem to work okay.

maddionatoplus, 2022-02-01:
How can I refer to that package directly from github?

ekuleshov, 2022-02-01:
> How can I refer to that package directly from github?

This is pointing to the latest commit. You can put any tag, branch or commit id into the `ref` value

```
  flutter_blue:
    git:
      url: https://github.com/espresso3389/flutter_blue
      ref: 9758034dc41956d7edd3b6917f3ed3dd9eab5ef1
```

maddionatoplus, 2022-02-03:
I receive errors with the compileSdkVersion:

An exception has occurred in the compiler (1.8.0_242-release). Please file a bug against the Java compiler via the Java bug reporting page (http://bugreport.java.com) after checking the Bug Database (http://bugs.java.com) for duplicates. Include your program and the following diagnostic in your report. Thank you.      
java.lang.AssertionError: annotationType(): unrecognized Attribute name MODULE (class com.sun.tools.javac.util.UnsharedNameTable$NameImpl)

Actually I have this:

android {
    compileSdkVersion flutter.compileSdkVersion
    ...

If I put 30 I receive error on BLE permissions

gibbsvjy007, 2022-03-25:
> > How can I refer to that package directly from github?
> 
> This is pointing to the latest commit. You can put any tag, branch or commit id into the `ref` value
> 
> ```
>   flutter_blue:
>     git:
>       url: https://github.com/espresso3389/flutter_blue
>       ref: 9758034dc41956d7edd3b6917f3ed3dd9eab5ef1
> ```

this solution worked for me, do you know when it will be merged to official plugin ?

don41382, 2022-04-03:
I was already close to discarding my ble app idea in flutter, but you saved the day. Thanks @espresso3389 🎉 

Would be really nice, if it would be merged into the offical plugin.

vricosti, 2022-04-14:
Will the fork be officially merged one day ?

Kampfotter, 2022-04-19:
Yes when will it be merged! This is essential for our app.

HanJiuYuan, 2022-05-06:
Hello, I would like to know how to solve the error when I pack APK
[image]
.

boskokg, 2022-05-06:
@HanJiuYuan you should use flutter_blue_plus where we continued work on this plugin.

HanJiuYuan, 2022-05-07:
Hello, I updated this plug-in, but there are still error
[image]
s

HanJiuYuan, 2022-05-07:
@boskokg Hello, I updated this plug-in, but there are still error
[image]
s

HanJiuYuan, 2022-05-07:
@boskokg Hello, I updated this plug-in, but there are still errors
[image]

boskokg, 2022-05-07:
@HanJiuYuan please for further question on flutter_blue_plus open issue there (not here).
I am not sure what is the java version that you use, but is seems that you are using some old version.

HanJiuYuan, 2022-05-07:
> 

@boskokg The Version of Java I am currently using is 1.8
[image]

tomasatisocco, 2022-10-13:
> My own flutter_blue fork that supports Android 12 permission: https://github.com/espresso3389/flutter_blue
> 
> It's still under development but the example code seems work correctly. I'm preparing PR but the changes are relatively large and I'm not sure it can be safely merged or not...

Using your fork solve my problems with Android 12 thank you!

bharatbudhori, 2022-11-19:
ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(startScan, Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@fdbd338: GattService registerScanner, java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@fdbd338: GattService registerScanner

I am getting this error while trying to scan for devices, through flutter_blue package

MrCsabaToth, 2022-11-19:
> ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(startScan, Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@fdbd338: [...]

`android.permission.BLUETOOTH_SCAN` and `android.permission.BLUETOOTH_CONNECT` are new permissions related to Android 12. Consequently you don't need location permissions any more for Bluetooth. I assume that you target SDK level 31 (Android 12) or above (32 is Android 12L) so Android applies newer rules. The last commit to `flutter_blue` was in March 2021, so I advise to switch to https://pub.dev/packages/flutter_blue_plus (which is a community maintained fork of flutter_blue) and try if that solves the issue.

dirtyhornet277, 2023-03-26:
> My own flutter_blue fork that supports Android 12 permission: https://github.com/espresso3389/flutter_blue
> 
> It's still under development but the example code seems work correctly. I'm preparing PR but the changes are relatively large and I'm not sure it can be safely merged or not...

Thanks @espresso3389, you help me! +REP!

rodneyosodo, 2023-11-05:
Thanks @espresso3389 for this. I had the same issue, solved it by using your fork 😄. I would appreciate if this fix could be merged on the main branch of `flutter_blue`

sudipmahanta, 2023-11-22:
I am using flutter_blue: ^0.8.0 and its working fine.
In first just check the permission for bluetoothScan isGranted or not using permission_handler

Kamal3113, 2024-02-09:
kindly , add this permission check before scan:
await Permission.bluetoothConnect.request().isGranted;

this will help in each and every (Android Version).