pauldemarco/flutter_blue #865: Prevent Andriod/iOS from pairing attempt

xritzx, 2021-06-08:
In my scenario, I want to connect and write some data into a BLE characteristic and disconnect from the device, this is achieved smoothly without even pairing the device even if I cancel the pairing request.
Is there is a way to prevent the OS from popping up with a pair request dialog? since I can write into a given characteristic and disconnect I don't even want to pair with the device

AlainHsu, 2021-06-10:
As far as i know, it can be avoided from firmware side by not trigger pairing or not encrypting characteristic.

xritzx, 2021-06-10:
> As far as i know, it can be avoided from firmware side by not trigger pairing or not encrypting characteristic.

Solved:
Hi, thanks for the reply, yes the problem was with the hardware's firmware as it was asking for passkey authentication, I had to modify that code and remove the passkey paring and it solved the problem I was facing,

closed 2021-06-10 by xritzx