pauldemarco/flutter_blue #870: macOS: empty string service UUID causing bad format exception

meros, 2021-06-16:
I have a device whose services are for some reason marked as not primary on macOS (but not iOS). There are however no other services including those services, which causes findPrimaryService to return nil and causing serviceUuid in the proto buffer object to become "".

I understand that it's not correct for the device to return secondary devices without a primary device, but I think we need a fallback behaviour in these cases...

[image]
[image]

meros, 2021-06-16:
After some more testing this seems to be a more general problem that isPrimary is NO for any service on macOS, but not iOS. I have tried to find more information around this to no avail

Mirageofmage, 2021-07-23:
#657