pauldemarco/flutter_blue #873: Fundamental issue regarding accessing characteristics (and more?) with same uuid

meros, 2021-06-17:
It is allowed to have multiple characteristics with the same UUID, in fact TWS headphones use this fact to expose multiple battery values. The different characteristics can be discerned using the descriptor, but as it is now this library is fundamentally unable to handle this situation.

We need to access services, characteristics and descriptors not by their UUID but rather some internal ID (like we do for peripherals)

mgeilich, 2023-02-28:
I am having this same problem.  Any progress or alternative?  I tried the flutter_reactive_ble plugin and it has the same issue.   I can retrieve two characteristics with the same UUID from services.characteristics, but all writes seem to go to the first of the two.

meros, 2023-02-28:
I don't remember what I did in the end. I might have made a custom solution. Sadly I can't help you more than that. Good luck with your characteristics and take care out there 🫡