pauldemarco/flutter_blue #875: set_notification_error, could not locate CCCD descriptor for characteristic

jason-juntong, 2021-06-19:
### Could not locate CCCD descriptor for characteristic

When tried to execute `await characteristic.setNotifyValue(true);` will cause an error. 

Error: 
`E/flutter (28172): [ERROR:flutter/lib/ui/ui_dart_state.cc(186)] Unhandled Exception: PlatformException(set_notification_error, could not locate CCCD descriptor for characteristic: 00002a00-0000-1000-8000-00805f9b34fb, null, null)`

The code bellow is the whole operation from discovering services and read characteristic.

`discoverServices() async {
List<BluetoothService> services = await discoveredDevice.discoverServices();
services.forEach((service) {
    service.characteristics.forEach((characteristic) async {

        // Read characteristic
        var value = await characteristic.read();
        debugPrint("Read value: $value");

        // Read descriptor
        var descriptors = characteristic.descriptors;
        for(BluetoothDescriptor d in descriptors) {
          List<int> descriptorValue = await d.read();
          debugPrint("Descriptor value: $descriptorValue");
        }

        await characteristic.setNotifyValue(true);
        debugPrint('Is characteristic notifying: ${characteristic.isNotifying}');
        characteristic.value.listen((characteristicValue) {
          debugPrint('Characteristic value: $characteristicValue');
          setState(() {
            listStream = characteristic.value;
          });
        });
    });
});`

yuriboeira11tx, 2021-06-22:
Same problem.

RazaGR, 2021-06-28:
no solution?

JKornberg, 2021-07-06:
Have the same problem, found this article: https://www.programmersought.com/article/17006030580/

Note: my app works great on iOS, this only happens on Android

JamesKarl, 2022-02-24:
I fixed this issue by modifying the plugin source code as below. 

**flutter_blue-0.8.0/android/src/main/java/com/pauldemarco/flutter_blue/FlutterBluePlugin.java**

**line start at 496**
```java
 //BluetoothGattDescriptor cccDescriptor;
                try {
                    gattServer = locateGatt(request.getRemoteId());
                    characteristic = locateCharacteristic(gattServer, request.getServiceUuid(), request.getSecondaryServiceUuid(), request.getCharacteristicUuid());
                   /* cccDescriptor = characteristic.getDescriptor(CCCD_ID);
                    if(cccDescriptor == null) {
                        throw new Exception("could not locate CCCD descriptor for characteristic: " +characteristic.getUuid().toString());
                    }*/
                } catch(Exception e) {
                    result.error("set_notification_error", e.getMessage(), null);
                    return;
                }
```
**line start at 570**
```java
/* if(!cccDescriptor.setValue(value)) {
                    result.error("set_notification_error", "error when setting the descriptor value to: " + value, null);
                    return;
                }

                if(!gattServer.writeDescriptor(cccDescriptor)) {
                    result.error("set_notification_error", "error when writing the descriptor", null);
                    return;
                }*/
```

WollmannDaniel, 2022-09-14:
Any news on this topic?

v4542, 2024-07-26:
same problem,  i tried running the app on two devices, and one got the app running and the other stucked here

WollmannDaniel, 2024-07-26:
> same problem,  i tried running the app on two devices, and one got the app running and the other stucked here 
> 
> 

Try to use package flutter_blue_plus. Seems to fix some of the errors in this package.