pauldemarco/flutter_blue #895: iOS scan in background

meridorkobi, 2021-07-16:
iOS in background can do only passive scanning.
How can I do it with flutter_blue. 
following listener code

```
            flutterBlueInstance.scan(scanMode: ScanMode.lowPower ).listen((event) {
              if ( event.advertisementData.localName == "SHS-FAA4") {
                  final DateTime now = DateTime.now();
                  loggerDBNDI("iiii found in listen [$now] -" + event.device.name.toString());
                }
              }
            )

```

oddko, 2021-07-17:
Hi, as far as iOS is concerned, you need to specify a list of service UUIDs ('withServices' argument of https://pub.dev/documentation/flutter_blue/latest/flutter_blue/FlutterBlue/scan.html) to be able to let the scan run in the background (see Apple documentation https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518986-scanforperipherals )

meridorkobi, 2021-07-17:
Add the Guid to scan command
flutterBlueInstance.scan( withServices: [Guid("XXXXXXXuuidXXXXXXX")] ).listen((

closed 2021-07-17 by meridorkobi

meridorkobi, 2021-07-20:
I have a beacon and can't config the GUID. 
Do we have other option for iOS in background?

oddko, 2021-07-21:
Unfortunately not, if you wanna scan for any bluetooth peripheral on iOS the user will have to keep the application running in foreground.