pauldemarco/flutter_blue #900: Empty scan results when screen is off

koenniem, 2021-07-22:
## Motivation
I want to run a Bluetooth scan every 20 or so seconds, even with the screen off. In my actual application, this is achieved by running this scan from a background isolate, though naturally it should work while the app is on the foreground is as well. Thus, while keeping the app open, Bluetooth scans should keep working even when you toggle the screen off.

## Issue
A background scan is performed, but no Bluetooth devices are found (i.e. empty `FlutterBlue.instance.scanResults`).

## Steps to reproduce
Download the example app and library from this Github.

Make sure to ask for the proper permissions in `AndroidManifest.xml` 
```XML
<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
```

Schedule the background scan immediately at build in `main.dart`
```dart 
class FlutterBlueApp extends StatelessWidget {
  @override
  Widget build(BuildContext context) {
    ScreenOff().backgroundScan();
    return MaterialApp(
      ...
    );
  }
}
``` 

And then add the `ScreenOff` class in `main.dart` to start the timer for the scan
```dart
class ScreenOff {
  static final ScreenOff _singleton = ScreenOff._internal();

  factory ScreenOff() {
    return _singleton;
  }

  ScreenOff._internal();

  Future<void> backgroundScan() async {
    new Timer.periodic(const Duration(seconds: 20), (Timer t) => {
      FlutterBlue.instance.startScan(timeout: Duration(seconds: 5))
    });

    // Listen to stream and print to console
    FlutterBlue.instance.scanResults.forEach(print);
  }
}
```

## Notes
Tested on a Xiaomi Mi 10T Lite (Android 11, MIUI 12) and Samsung S8 (Android 9).

I've already checked if this could be related to #210 or #191 but it seems `FlutterBluePlugin` already registers with `context` instead of `activity` which is the proposed solution in these PRs. Any ideas what I'm missing or is this simply not implemented yet?

oddko, 2021-07-26:
Why would you make the scan to time out after 5 seconds ? Just let it run indefinitely, scan results are emitted as they arrive, not once the scan stops. By letting it only run for 5 seconds every 20 seconds it's likely that it does not even have the time to properly receive and process advertisements.

koenniem, 2021-07-29:
I do not want to leave the scan running indefinitely because of unnecessary battery drain. For my application, advertisements aren't every necessary as I'm only interested in the MAC address. 

Though this is all beside the point for the current discussion. What I really want is to have Bluetooth scan in the background, continuous or not.