pauldemarco/flutter_blue #902: Unable to write to characteristic in release version of app (even following #331 and #772)

anthonyorona, 2021-07-24:
Issue #331 describes this but I have not found it to be a solution for my current app

These are my android directories:

android/app/build.gradle

...
```
android {
    compileSdkVersion 30

    defaultConfig {
        applicationId "com.mycompany.bluetooth"
        minSdkVersion 19
        targetSdkVersion 28
        versionCode flutterVersionCode.toInteger()
        versionName flutterVersionName
        multiDexEnabled true
    }

    buildTypes {
        release {
            signingConfig signingConfigs.debug
            shrinkResources false
            minifyEnabled false
            useProguard true
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
}
```

android/app/proguard-rules.pro

- proguard-rules.pro

    -ignorewarnings
    -keep class com.pauldemarco.flutter_blue.* { *; }
    -keep class com.pauldemarco.flutter_blue.Protos* { *; }

--- As far as I can tell from Issue #772 and #331 I can do this and then the release version should work.

In addition this is how I use the library:

First I set up the service and characteristics
```
    // 1. Discover Service
    List<BluetoothService> services = await device.discoverServices();

    // 2. Get Service
    var myService = services.firstWhere((service) => service.uuid.toString() == '<my-service-id>');

    // 3. Get Characteristics
    for (BluetoothCharacteristic characteristic in characteristics) {
      if (characteristic.uuid.toString() == readCharacteristicID) {
        _readCharacteristic = characteristic;
      }
      if (characteristic.uuid.toString() == writeCharacteristicID) {
        _writeCharacteristic = characteristic;
      }
    }

    await _readCharacteristic.setNotifyValue(true);
    _listenerSubscription = _readCharacteristic.value.listen(() {
       // my basic listener
    });
   await device.requestMtu(250); // Max MTU on Android is 517 bytes BTW
```

Then I call a function asynchronously with a lock to write the characteristic. The lock is there so that serial access to other asynchronous methods in the body (that I have omitted) are accessed exclusively.
```
  Future<bool> _makeSerialHLRequest(Uint8List data) async {
    await _lock.synchronized(() async {
      await _writeCharacteristic.write(datagram);
    }
  }
```

I have tried many different means of getting the write characteristic to work.

If I do await _writeCharacteristic.write(datagram); await Future.wait([_writeCharacteristic.write(datagram)]) or  _writeCharacteristic.write(datagram); then it works when on adb but not when in release. If I add a timeout directly to the write characteristic or add a timeout to the Future.wait call it also fails.

In every case when I am in release I will get an Exception thrown from the platform:

```
I/flutter (27954): PlatformException(write_characteristic_error, writeCharacteristic failed, null, null), #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:597)
I/flutter (27954): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:158)
I/flutter (27954): <asynchronous suspension>
I/flutter (27954): #2      BluetoothCharacteristic.write (package:flutter_blue/src/bluetooth_characteristic.dart:120)
I/flutter (27954): <asynchronous suspension>
I/flutter (27954): #3      Future.wait.<anonymous closure> (dart:async/future.dart)
I/flutter (27954): <asynchronous suspension>
I/flutter (27954): #4      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:813)
I/flutter (27954): <asynchronous suspension>
```

This was working before, now it is not working. I have tried building as follows:

    flutter build apk --no-sound-null-safety --no-shrink

This is a pretty critical piece of the library, it seems like there is an underlying issue with the plugin and how it works with the Android bluetooth stack if this is such a persistent issue across multiple issues and versions of the library. I hope that the underlying cause can be understood and addressed.  I would really like to be told I am using the library incorrectly so that I can just fix the issue and move on, but this seems like a pretty nasty bug as it pops up on release but not when tethered (and it doesn't have a very helpful stacktrace).

anthonyorona, 2021-07-24:
Ya I was playing around with calling the write characteristic and discovered that if I pause 1 second between requesting MTU set and writing the characteristic, it all works...

So...figure this one out.

```
      await device.requestMtu(250); // Max MTU on Android is 517 bytes BTW
      await Future.delayed(Duration(seconds: 1));
```

So, what is happening in the library where pausing after requesting MTU change is necessary for writing after? the collision issue?

oddko, 2021-07-26:
Had the same issue back then, it is due to the fact that the call to set the MTU returns in non-blocking fashion, before the new MTU actually got set/negociated on the bluetooth transport layer. The way I ended up doing it is listening for changes in the MTU and waiting for said requested value (e.g. 250) to be received in the 'mtu' property. Simply use a Completer that you'll await on after chaning the MTU, that is completed when the mtu changes to the desired value.

anthonyorona, 2021-07-26:
Thank you @oddko 

Aha, for anyone coming across this in the future you can simply do:

    await device.requestMtu(250); // I would await this regardless, set a timeout if you are concerned

    var mtuChanged = Completer<void>();

    // mtu is of type 'int'
    var mtuStreamSubscription = device.mtu.listen((mtu) {
        if(mtu == 250) mtuChanged.complete();
    });
    await awaitMtuChanged.future; // set timeout and catch exception
    mtuStreamSubscription.cancel();

You may want to set a timeout on the future and catch the exception and retry up to x times just to avoid hanging on this.

I have noticed there are several issues logged that deal with a similar problem. The expectation for many would be that once the future completes for requestMtu(<int>) the MTU would be set and it would be safe to proceed. If that is not the case we should at least note it for people in the future.

#476 
#447 
#579
#800

closed 2021-07-26 by anthonyorona