pauldemarco/flutter_blue #907: Could not resolve all files for configuration ':flutter_blue:protobufToolsLocator_protoc' Apple M1 chip

JAICHANGPARK, 2021-07-29:
```
Execution failed for task ':flutter_blue:generateDebugProto'.
> Could not resolve all files for configuration ':flutter_blue:protobufToolsLocator_protoc'.
   > Could not find protoc-3.11.4-osx-aarch_64.exe (com.google.protobuf:protoc:3.11.4).
     Searched in the following locations:
         https://jcenter.bintray.com/com/google/protobuf/protoc/3.11.4/protoc-3.11.4-osx-aarch_64.exe
```

Pesta666, 2021-07-30:
Same here on M1 and flutter_blue v0.8.0

vborbely, 2021-08-04:
> Same here on M1 and flutter_blue v0.8.0, anyone solved?

I created this comment as a solution:

https://github.com/protocolbuffers/protobuf/issues/8062#issuecomment-892689597

I hope this helps!

Good luck!

teddichiiwa, 2021-08-19:
@vborbely it's working perfectly now.

sultan18kh, 2021-08-25:
Hey @vborbely, I still wasn't able to solve this problem for Android for my M1 Mac. Any update on the latest version of macOS Big Sur 11.5.2?

boskokg, 2022-02-13:
use flutter_blue_plus

tenutos, 2022-09-01:
Please update the protobuf section in build.gradle to
import org.apache.tools.ant.taskdefs.condition.Os
// Compatible with macOS on Apple Silicon
def archSuffix = Os.isFamily(Os.FAMILY_MAC) ? ':osx-x86_64' : ''
protobuf {
    protoc {
        artifact = "com.google.protobuf:protoc:3.20.1$archSuffix"
    }
    plugins {
        javalite {
            artifact = "com.google.protobuf:protoc-gen-javalite:3.0.0$archSuffix"
        }
    }
}
Since you are now locked out of artifact = "com.google.protobuf:protoc:3.11.4", I specified in my own project that artifact = "com.google.protobuf:protoc:3.11.4:osx-x86_64" is invalid, which caused me to go to flutter_blue's source code and change it manually, which was really annoying.
My development environment is macOS on Apple Silicon.