pauldemarco/flutter_blue #909: UART BLE Simulation?

laguierre, 2021-07-31:
Hi, I need to simulate Nordic UART over BLE. The only way I found it's using the hardware. However it would be useful to simulate this on the Android Emultator. 
So when you, for example, start FlutterBlue.instance.startScan... only appears beacons. Is it possible to make and Nordic UART and Battery Service too? 
SERVICE_UUID = "6E400001-B5A3-F393-E0A9-E50E24DCCA9E";  
CHARACTERISTIC_UUID_RX = "6E400003-B5A3-F393-E0A9-E50E24DCCA9E";
CHARACTERISTIC_UUID_TX = "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"; 
Thank

sjrdvndwg, 2021-11-02:
i dont think this library supports it but this [one](https://github.com/dotintent/FlutterBleLib) does when combined with this [one](https://pub.dev/packages/blemulator)

laguierre, 2021-11-11:
Thank! Sounds interesting, I will try. I am new in Flutter, it's not my core, I am electronic. So I used Flutter_Ble becuase worked and it was more intuitive. 
However I suppose the best idea is go throught Polidea...
Do you make some with BLE?

closed 2021-11-11 by laguierre