pauldemarco/flutter_blue #912: void Function(BluetoothDeviceState) callback in FlutterBlue.instance.scan().listen(callback) not invoked for BluetoothDeviceState.connecting

ReubenBeeler, 2021-08-05:
I am trying to determine when a device is connecting so I can update the UI, but the `BluetoothDevice#state` stream returned by `FlutterBlue.instance.scan()` does not hear `BluetoothDeviceState.connecting`.

When I add a listener to a device, it listens to the initial `BluetoothDeviceState.disconnected`. After calling `BluetoothDevice#connect()` (and waiting a second or so), the stream hears `BluetoothDeviceState.connected`, without any intermediate `BluetoothDeviceState.connecting`. A workaround is setting some local `bool isConnecting` to true before calling `BluetoothDevice#connect()` and then to false after awaiting, but it is somewhat annoying.

If there is a non-bug reason for this, please let me know.

vardelean, 2021-08-18:
I second that, I just noticed the same behavior, I'd love to be able to use the interim states connecting and disconnecting. 

Using a flag (i.e. bool isConnecting) is what I was thinking as well, but I'm not a big fan of working with flags. My solution was to start a circular progress indicator when the connection request starts and stop it when the device is connected, and similar when disconnecting the device.

Also, perhaps this is not a bug (missing the connecting and disconnecting states), but just a feature to be implemented at one point.

anthonyorona, 2021-08-23:
connecting and disconnecting are state transitions I would just use flags. It is a bit odd though isn't it? That they've included those as states but they don't come up in the stream.

Philippe-Nau, 2021-09-30:
I also need to use connecting and disconnecting, but they are not working.