pauldemarco/flutter_blue #915: scanning... Bluetooth device name / advertisementData.localName may be incorrect and actually be the name of a different device...

dspells, 2021-08-10:
This problem is for iOS only. I'm testing on iOS 14.0.1. Android gets the names correctly. Here is an example

2021-08-10 18:15:45 +0000 SwitchDiscovery.setScanResults/D:	ScanResult{device: BluetoothDevice{id: F72DE1B4-8E13-6EA3-475E-F8D93B6DF141, name: BlueZ 5.50, type: BluetoothDeviceType.le, isDiscoveringServices: false, _services: [], advertisementData: AdvertisementData{localName: AP-Texas-QA-4K-1.08-801, txPowerLevel: null, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: []}, rssi: -65})

Note: The device name is BlueZ 5.50 and the  AdvertisementData localName is AP-Texas-QA-4K-1.08-801.

This is the debug output from Android.

ScanResult{device: BluetoothDevice{id: 34:C9:F0:83:16:04, name: AP-Texas-QA-4K-1.08-801, type: BluetoothDeviceType.le, isDiscoveringServices: false, _services: [], advertisementData: AdvertisementData{localName: AP-Texas-QA-4K-1.08-801, txPowerLevel: null, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: []}, rssi: -68}

Note: Both the device name and the  AdvertisementData localName is AP-Texas-QA-4K-1.08-801.

We look at the device name to determine if it might one our network packet broker switches advertising over BLE.

Flutter 2.2.3 • channel stable • https://github.com/flutter/flutter.git
Framework • revision f4abaa0735 (6 weeks ago) • 2021-07-01 12:46:11 -0700
Engine • revision 241c87ad80
Tools • Dart 2.13.4

  flutter_blue:
    dependency: "direct main"
    description:
      name: flutter_blue
      url: "https://pub.dartlang.org"
    source: hosted
    version: "0.8.0"

Here is snippet of how that output was created
```
  void setScanResults(AsyncSnapshot<List<ScanResult>> snapshot) {
    logger.d('enter ${snapshot.data?.length ?? 0} async $_inAsyncCall');
    for (ScanResult r in results) {
      if (r.device.name.isEmpty && r.advertisementData.localName.isNotEmpty) logger.e(r.toString());
      if (r.advertisementData.localName.startsWith('AP-')) logger.d(r.toString());
    }
  }
```

I will post more detailed ScanResults lists in two separate files (iOS and Android).

dspells, 2021-08-10:
[android scan.txt](https://github.com/pauldemarco/flutter_blue/files/6963847/android.scan.txt)
This is the results of the scan on android

dspells, 2021-08-10:
[ios scan 1.txt](https://github.com/pauldemarco/flutter_blue/files/6963877/ios.scan.1.txt)
This is the results of a bad scan on iOS. The problem is intermittent (Maybe 50/50).

dspells, 2021-08-10:
[android scan.txt](https://github.com/pauldemarco/flutter_blue/files/6963937/android.scan.txt)
This is the output from android.

dspells, 2021-08-10:
[ios scan 1.txt](https://github.com/pauldemarco/flutter_blue/files/6963961/ios.scan.1.txt)
This is the results of a bad scan on iOS.

dspells, 2021-08-10:
Why does Android get the correct name but not iOS?

dspells, 2021-08-10:
Is this the same as https://github.com/pauldemarco/flutter_blue/issues/524?

dspells, 2021-08-19:
As a work around I am only using the advertisement localName instead of the device name to recognize our switches and show that name in our mobile app GUI. I am still clueless as to why the device name is what we expect on Android and not what we expect on iOS.

bruce-qingwu, 2022-02-17:
Is there any update about this issue? For a BLE device (esp32) I am using, the advertismentData.localName is always empty and device name cannot show properly the first time on the iOS.

Mohitkau, 2022-05-05:
@bruce-qingwu, same with me. I found the initial scanning result is name is empty and might be taking some milliseconds to fetch the name. This works for me you can check.

https://github.com/pauldemarco/flutter_blue/issues/524#issuecomment-1118866250