pauldemarco/flutter_blue #918: iOS only BluetoothCharacteristic.read returns munged / duplicated data when data is 'large'. Android works perfectly.

dspells, 2021-08-19:
Here is my code to read a bluetooth characteristic

```
  Future<VT> read() async {
    logger.d('$name');
    assert(null != characteristic);
    if (null == characteristic) return Future.error(Exception(Strings.lang.kNullBluetoothCharacteristic));
    Mutex m = AppState.instance.connectedSwitch.aBluetoothMutex;

    logger.d('protect');
    return m.protect<VT>(() async {
      return await characteristic!.read().then((values) {
        logger.i('$name len ${values.length}');
        _value.update = converter.fromInts(values);
        logger.d('end protect value ${current.toString()}');
        return original;
      });
    });
  }
```

On Android I get the correct response

```I/Characteristic.read.<ac>.<ac>( 2031): cliResponse0 len 489```
```D/Characteristic.read.<ac>.<ac>( 2031): end protect value [120, 218, 205, 84, 219, 110, 211, 64, 16, 253, 149, 209, 62, 129, 228, 34, 219, 73, 227, 216, 60, 69, 129, 162, 74, 180, 66, 73, 169, 4, 85, 31, 54, 246, 36, 89, 186, 23, 179, 23, 66, 84, 229, 223, 25, 55, 105, 156, 11, 80, 137, 155, 120, 90, 237, 236, 156, 179, 51, 231, 204, 238, 61, 27, 72, 180, 222, 177, 226, 230, 158, 157, 87, 172, 72, 178, 78, 154, 231, 17, 187, 90, 214, 200, 10, 54, 14, 19, 83, 123, 161, 184, 132, 129, 176, 83, 105, 22, 44, 98, 151, 92, 53, 103, 195, 57, 119, 78, 56, 10, 140, 61, 247, 77, 196, 212, 168, 155, 45, 126, 65, 43, 252, 146, 34, 74, 104, 99, 41, 244, 10, 93, 105, 5, 49, 25, 221, 34, 129, 22, 72, 1, 85, 237, 151, 224, 164, 241, 238, 37, 156, 241, 137, 21, 229, 144, 219, 234, 153, 123, 14, 31, 34, 248, 72, 232, 43, 161, 208, 121, 174, 106, 194, 166, 113, 154, 156, 196, 253, 147, 36, 135, 164, 83, 116, 210, 162, 19, 83, 198, 27, 107, 66, 189, 87, 211, 5, 119, 119, 66, 207, 198, 38, 216, 178, 169, 173, 220, 158, 12, 202, 59, 164, 78, 167, 92, 58, 124, 216, 17, 72, 207, 118, 66, 13, 116, 103, 59, 148, 198, 33, 159, 72, 108, 19, 76, 117, 174, 43, 252, 74, 114, 229, 253, 126, 150, 173, 162, 86, 188, 126, 182, 21, 239, 29, 183, 94, 163, 133, 161, 209, 222, 26, 73, 66, 195, 120, 169, 203, 86, 193, 183, 56, 245, 59, 167, 63, 87, 146, 127, 250, 142, 146, 7, 12, 240, 120, 101, 185, 110, 9, 166, 194, 170, 5, 183, 8, 196, 227, 26, 200, 19, 98, 198, 217, 63, 23, 147, 74, 230, 106, 61, 128, 27, 89, 74, 30, 28, 18, 247, 53, 151, 97, 87, 199, 31, 54, 181, 186, 61, 242, 164, 183, 231, 73, 59, 208, 103, 143, 216, 235, 53, 22, 46, 132, 83, 220, 151, 243, 214, 149, 145, 152, 205, 127, 211, 150, 67, 10, 208, 136, 149, 59, 42, 188, 128, 228, 69, 156, 195, 36, 8, 89, 65, 158, 247, 226, 39, 237, 73, 254, 3, 123, 54, 197, 143, 240, 115, 16, 150, 176, 173, 81, 135, 221, 28, 251, 210, 59, 221, 243, 229, 180, 125, 43, 102, 209, 60, 143, 80, 215, 114, 9, 239, 117, 45, 195, 108, 246, 192, 189, 185, 180, 251, 11, 46, 236, 113, 118, 33, 236, 176, 254, 9, 149, 187, 127, 225, 47, 233, 197, 141, 102, 175, 173, 53, 118, 104, 42, 202, 138, 35, 54, 66, 23, 164, 111, 166, 10, 189, 21, 212, 116, 69, 205, 112, 201, 173, 162, 47, 146, 109, 179, 101, 80, 250, 50, 168, 9, 13, 108, 17, 175, 190, 1, 94, 13, 225, 181]```

on iOS I get the munged data

```2021-08-19 17:38:13 +0000 Characteristic.read.<ac>.<ac>/I:	cliResponse0 len 552)```
```2021-08-19 17:38:13 +0000 Characteristic.read.<ac>.<ac>/D:	end protect value [120, 218, 205, 84, 219, 110, 211, 64, 16, 253, 149, 209, 62, 129, 228, 34, 219, 73, 227, 216, 60, 69, 129, 162, 74, 180, 66, 73, 169, 4, 85, 31, 54, 246, 36, 89, 186, 23, 179, 23, 66, 84, 229, 223, 25, 55, 105, 156, 11, 80, 137, 155, 120, 90, 237, 236, 156, 179, 51, 231, 204, 238, 61, 27, 72, 180, 222, 177, 226, 230, 158, 157, 87, 172, 72, 178, 78, 154, 231, 17, 187, 90, 214, 200, 10, 54, 14, 19, 83, 123, 161, 184, 132, 129, 176, 83, 105, 22, 44, 98, 151, 92, 53, 103, 195, 57, 119, 78, 56, 10, 140, 61, 247, 77, 196, 212, 168, 155, 45, 126, 65, 43, 252, 146, 34, 74, 104, 99, 41, 244, 10, 93, 105, 5, 49, 25, 221, 34, 129, 22, 72, 1, 85, 237, 151, 224, 164, 241, 238, 37, 156, 241, 137, 21, 229, 144, 219, 234, 153, 123, 14, 31, 34, 248, 72, 232, 43, 161, 208, 121, 174, 106, 194, 166, 113, 154, 156, 196, 253, 147, 36, 135, 164, 83, 120, 218, 205, 84, 219, 110, 211, 64, 16, 253, 149, 209, 62, 129, 228, 34, 219, 73, 227, 216, 60, 69, 129, 162, 74, 180, 66, 73, 169, 4, 85, 31, 54, 246, 36, 89, 186, 23, 179, 23, 66, 84, 229, 223, 25, 55, 105, 156, 11, 80, 137, 155, 120, 90, 237, 236, 156, 179, 51, 231, 204, 238, 61, 27, 72, 180, 222, 177, 226, 230, 158, 157, 87, 172, 72, 178, 78, 154, 231, 17, 187, 90, 214, 200, 10, 54, 14, 19, 83, 123, 161, 184, 132, 129, 176, 83, 105, 22, 44, 98, 151, 92, 53, 103, 195, 57, 119, 78, 56, 10, 140, 61, 247, 77, 196, 212, 168, 155, 45, 126, 65, 43, 252, 146, 34, 74, 104, 99, 41, 244, 10, 93, 105, 5, 49, 25, 221, 34, 129, 22, 72, 1, 85, 237, 151, 224, 164, 241, 238, 37, 156, 241, 137, 21, 229, 144, 219, 234, 153, 123, 14, 31, 34, 248, 72, 232, 43, 161, 208, 121, 174, 106, 194, 166, 113, 154, 156, 196, 253, 147, 36, 135, 164, 83, 120, 218, 205, 84, 219, 110, 211, 64, 16, 253, 149, 209, 62, 129, 228, 34, 219, 73, 227, 216, 60, 69, 129, 162, 74, 180, 66, 73, 169, 4, 85, 31, 54, 246, 36, 89, 186, 23, 179, 23, 66, 84, 229, 223, 25, 55, 105, 156, 11, 80, 137, 155, 120, 90, 237, 236, 156, 179, 51, 231, 204, 238, 61, 27, 72, 180, 222, 177, 226, 230, 158, 157, 87, 172, 72, 178, 78, 154, 231, 17, 187, 90, 214, 200, 10, 54, 14, 19, 83, 123, 161, 184, 132, 129, 176, 83, 105, 22, 44, 98, 151, 92, 53, 103, 195, 57, 119, 78, 56, 10, 140, 61, 247, 77, 196, 212, 168, 155, 45, 126, 65, 43, 252, 146, 34, 74, 104, 99, 41, 244, 10, 93, 105, 5, 49, 25, 221, 34, 129, 22, 72, 1, 85, 237, 151, 224, 164, 241, 238, 37, 156, 241, 137, 21, 229, 144, 219, 234, 153, 123, 14, 31, 34, 248, 72, 232, 43, 161, 208, 121, 174, 106, 194, 166, 113, 154, 156, 196, 253, 147, 36, 135, 164, 83])```

After spending time analyzing the response visually I noticed that the first 184 bytes had been duplicated twice. Here is the above response broken out into three chunks.

```120, 218, 205, 84, 219, 110, 211, 64, 16, 253, 149, 209, 62, 129, 228, 34, 219, 73, 227, 216, 60, 69, 129, 162, 74, 180, 66, 73, 169, 4, 85, 31, 54, 246, 36, 89, 186, 23, 179, 23, 66, 84, 229, 223, 25, 55, 105, 156, 11, 80, 137, 155, 120, 90, 237, 236, 156, 179, 51, 231, 204, 238, 61, 27, 72, 180, 222, 177, 226, 230, 158, 157, 87, 172, 72, 178, 78, 154, 231, 17, 187, 90, 214, 200, 10, 54, 14, 19, 83, 123, 161, 184, 132, 129, 176, 83, 105, 22, 44, 98, 151, 92, 53, 103, 195, 57, 119, 78, 56, 10, 140, 61, 247, 77, 196, 212, 168, 155, 45, 126, 65, 43, 252, 146, 34, 74, 104, 99, 41, 244, 10, 93, 105, 5, 49, 25, 221, 34, 129, 22, 72, 1, 85, 237, 151, 224, 164, 241, 238, 37, 156, 241, 137, 21, 229, 144, 219, 234, 153, 123, 14, 31, 34, 248, 72, 232, 43, 161, 208, 121, 174, 106, 194, 166, 113, 154, 156, 196, 253, 147, 36, 135, 164, 83,```

```120, 218, 205, 84, 219, 110, 211, 64, 16, 253, 149, 209, 62, 129, 228, 34, 219, 73, 227, 216, 60, 69, 129, 162, 74, 180, 66, 73, 169, 4, 85, 31, 54, 246, 36, 89, 186, 23, 179, 23, 66, 84, 229, 223, 25, 55, 105, 156, 11, 80, 137, 155, 120, 90, 237, 236, 156, 179, 51, 231, 204, 238, 61, 27, 72, 180, 222, 177, 226, 230, 158, 157, 87, 172, 72, 178, 78, 154, 231, 17, 187, 90, 214, 200, 10, 54, 14, 19, 83, 123, 161, 184, 132, 129, 176, 83, 105, 22, 44, 98, 151, 92, 53, 103, 195, 57, 119, 78, 56, 10, 140, 61, 247, 77, 196, 212, 168, 155, 45, 126, 65, 43, 252, 146, 34, 74, 104, 99, 41, 244, 10, 93, 105, 5, 49, 25, 221, 34, 129, 22, 72, 1, 85, 237, 151, 224, 164, 241, 238, 37, 156, 241, 137, 21, 229, 144, 219, 234, 153, 123, 14, 31, 34, 248, 72, 232, 43, 161, 208, 121, 174, 106, 194, 166, 113, 154, 156, 196, 253, 147, 36, 135, 164, 83,```

```120, 218, 205, 84, 219, 110, 211, 64, 16, 253, 149, 209, 62, 129, 228, 34, 219, 73, 227, 216, 60, 69, 129, 162, 74, 180, 66, 73, 169, 4, 85, 31, 54, 246, 36, 89, 186, 23, 179, 23, 66, 84, 229, 223, 25, 55, 105, 156, 11, 80, 137, 155, 120, 90, 237, 236, 156, 179, 51, 231, 204, 238, 61, 27, 72, 180, 222, 177, 226, 230, 158, 157, 87, 172, 72, 178, 78, 154, 231, 17, 187, 90, 214, 200, 10, 54, 14, 19, 83, 123, 161, 184, 132, 129, 176, 83, 105, 22, 44, 98, 151, 92, 53, 103, 195, 57, 119, 78, 56, 10, 140, 61, 247, 77, 196, 212, 168, 155, 45, 126, 65, 43, 252, 146, 34, 74, 104, 99, 41, 244, 10, 93, 105, 5, 49, 25, 221, 34, 129, 22, 72, 1, 85, 237, 151, 224, 164, 241, 238, 37, 156, 241, 137, 21, 229, 144, 219, 234, 153, 123, 14, 31, 34, 248, 72, 232, 43, 161, 208, 121, 174, 106, 194, 166, 113, 154, 156, 196, 253, 147, 36, 135, 164, 83```

I added a print statement in BluetoothCharacteristic.read and the output is the same. I did this in attempt to verify that I was not munging the data somehow in my code.

At one point we were getting correct responses. I'm not sure where to go to debug this. Did I somehow get a new protobuf library that is broken? Did the iOS update from 14.0 to 14.0.1 and break this? etc...

Any clues on how to debug this would be appreciated.

igoriuz, 2021-08-20:
This has been reported multiple times. Try this PR: https://github.com/pauldemarco/flutter_blue/pull/850/files

#525
#840

dspells, 2021-08-20:
> This has been reported multiple times. Try this PR: https://github.com/pauldemarco/flutter_blue/pull/850/files
> 
> #525
> #840

That looks like a totally different bug. This involves reading in individual characteristic and getting duplicated data inside one read. The others look like reading streams of data. Do you still think it's the same?

dspells, 2021-08-20:
I added the following NSLog line into didUpdateValueForCharacteristic delegate method

```
- (void)peripheral:(CBPeripheral *)peripheral didUpdateValueForCharacteristic:(CBCharacteristic *)characteristic error:(NSError *)error {
  NSLog(@"didUpdateValueForCharacteristic after uuid: %@ value: %@", [characteristic.UUID fullUUIDString], [characteristic value]);
```

The returned value is 552 bytes when it should be 491 bytes. So as far as I can tell `readValueForCharacteristic` is sending the wrong data to the delegate method.

dspells, 2021-08-20:
I am closing out this issue. I currently believe that our hardware is incorrectly returning the wrong data if the mtu < 512 bytes. That is why is works on Android and not on iOS.

closed 2021-08-20 by dspells