pauldemarco/flutter_blue #923: Not Scanning Device List.Always Device size 0

Ravish1, 2021-08-29:
Hey I am using this plugin for my App.whenever it try to scan devices always provides the device size 0.I have location enabled as well during scan but nothing working.Please help me out any work around?

bbm3h18mpa, 2021-09-01:
I'm having the same/similar issue. I have set all the permissions but its not scanning. The weird thing about that is, that it works for my old android phone (lollipop android 5.0.1). Maybe it really has to do something about permissions because it seems to me that the old android versions are not so restricted with permissions.

cyclone-pk, 2021-09-01:
same issue I am unable to scan my le devices

Ravish1, 2021-09-05:
Same here I am testing on Android 11 and it didn't worked ,but when I tried on one of my friend device which is Samsung s8 and it worked .I have provided all the permission even manually but it didn't worked on my Device.

VirtualNonsense, 2021-09-19:
Same here. 
I'm testing on a Pixel 4a and it just refuses to work.

TAMILSELVAN943, 2021-11-08:
Same issue faced.When i scanning near by devices no devices displayed.Pls give me a solution

sasikdar, 2021-11-25:
I have a silly question but is it supposed to find another mobile device? can i can for other mobile devices with this library?