pauldemarco/flutter_blue #924: Android Gradle Plugin 7 integration

keehoo, 2021-08-30:
After upgrading android to android gradle plugin 7.1
`distributionUrl=https\://services.gradle.org/distributions/gradle-7.0.2-all.zip`

this happens:

`* What went wrong:
Some problems were found with the configuration of task ':flutter_blue:extractIncludeReleaseProto' (type 'ProtobufExtract').
  - Type 'com.google.protobuf.gradle.ProtobufExtract' property 'destDir' is missing an input or output annotation.

    Reason: A property without annotation isn't considered during up-to-date checking.

    Possible solutions:
      1. Add an input or output annotation.
      2. Mark it as @Internal.

    Please refer to https://docs.gradle.org/7.0.2/userguide/validation_problems.html#missing_annotation for more details about this problem.
  - Type 'com.google.protobuf.gradle.ProtobufExtract' property 'isTest' is missing an input or output annotation.

    Reason: A property without annotation isn't considered during up-to-date checking.

    Possible solutions:
      1. Add an input or output annotation.
      2. Mark it as @Internal.

    Please refer to https://docs.gradle.org/7.0.2/userguide/validation_problems.html#missing_annotation for more details about this problem.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org`

jonashoechst, 2021-12-06:
Have you come up with any fix / workaround for the issue?

AviFix, 2021-12-08:
Any solution?

jonashoechst, 2021-12-10:
We are currently using a fork which seems to be maintained more actively (https://github.com/boskokg/flutter_blue) and does work with Gradle 7.