pauldemarco/flutter_blue #925: Registration failed, unregister scannerId = 0

exp131, 2021-08-30:
I'm trying to run the example app on my Galaxy S9+, but every time I press Scan button I got this:
`
D/BluetoothAdapter(25359): STATE_ON
D/BluetoothLeScanner(25359): Start Scan with callback
D/BluetoothLeScanner(25359): onScannerRegistered() - status=133 scannerId=0 mScannerId=0
D/BluetoothLeScanner(25359): Registration failed, unregister scannerId = 0
D/BluetoothLeScanner(25359): Scan failed, reason: app registration failed
D/FlutterBluePlugin(25359): mDevices size: 0
D/BluetoothAdapter(25359): STATE_ON
D/BluetoothLeScanner(25359): could not find callback wrapper
D/FlutterBluePlugin(25359): mDevices size: 0
`
What can it be?

Berkant35, 2022-02-27:
Change package to _flutter blue plus_ **https://pub.dev/packages/flutter_blue_plus**

alholan, 2022-11-10:
**I have encountered same error with _https://pub.dev/packages/flutter_blue_plus_  :**
D/BluetoothLeScanner( 9732): Registration failed, unregister scannerId = 0
D/BluetoothLeScanner( 9732): Scan failed, reason: app registration failed

I restarted the phone and it works!