pauldemarco/flutter_blue #928: ios autoconnect doesnt work

compvisiondeveloper, 2021-09-01:
Hi,
ios auto connect doesn't work when I connect more than one devices. Could you help me for this? Is there any config in info.plist file.?

DanielPiflaks, 2021-09-22:
Hi, 
Is auto connect works for you on iOS system **when only one device is connected**?
For me, it works only on Android but on iOS it just never works.

boskokg, 2021-09-22:
I had the same issue. I wrote where is the bug and how to workaround it.

compvisiondeveloper, 2021-09-22:
@DanielPiflaks yes doesn't work auto connection  when only one device is connected  .

DanielPiflaks, 2021-09-22:
@compvisiondeveloper So it seems that auto connection just never works on iOS with flutter blue.
@boskokg For others to see your workaround in the other opened bug, here it is: https://github.com/pauldemarco/flutter_blue/issues/353

compvisiondeveloper, 2021-09-22:
Hi @DanielPiflaks ;
  I overlooked it your solution , i guess. Thanks for reply.

dhaval-parmar-1108, 2022-08-04:
Auto connect never works for IOS. Can't find any workaround yet.

FurkanTskn, 2022-08-11:
Hi @dhaval-parmar-1108 
We called a timer periodic function (per 10 second) that try connect to BLE device on start.
it's only solution that we have.