pauldemarco/flutter_blue #934: Some error occured when discovering services.

chenyilyang, 2021-09-14:
service uuid:49535343-fe7d-4ae5-8fa9-9fafd205e455
uuid: 49535343-1e4d-4bd9-ba61-23c647249616 value: (null)
uuid: 49535343-8841-43f4-a8d4-ecbe34729bb3 value: (null)
uuid: 49535343-6daa-4d02-abf6-19569aca69fe value: (null)
uuid: 49535343-aca3-481c-91ec-d85e28a60318 value: (null)
[ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: FormatException: The format is invalid
#0      Guid._fromString
package:flutter_blue/src/guid.dart:45
#1      new Guid
package:flutter_blue/src/guid.dart:15
#2      new BluetoothCharacteristic.fromProto
package:flutter_blue/src/bluetooth_characteristic.dart:35
#3      new BluetoothService.fromProto.<anonymous closure>
package:flutter_blue/src/bluetooth_service.dart:19
#4      MappedListIterable.elementAt (dart:_internal/iterable.dart:413:31)

chenyilyang, 2021-09-14:
On macos platform.

peekpt, 2021-09-15:
Did you try to remove the '-'  ?

rymca, 2021-11-10:
I have the same error on Bis Sure 11.6...