pauldemarco/flutter_blue #937: Write characteristic more than 20 bytes

HninNweAye, 2021-09-16:
Currently I can send only 20 bytes. I send more than 20 bytes then it will truncate all the data after 20 bytes.
I need to send large size of data. Is any one have this problem?

peekpt, 2021-09-16:
If you request MTU change you can send large chunks of data. But you can also split your data

HninNweAye, 2021-09-17:
I found someone mentioned this answer. It did solved my problem.

await bleReaderDevice?.requestMtu(250);

    final mtuChanged = Completer<void>();
    final mtuStreamSubscription = bleReaderDevice?.mtu.listen((mtu) {
      if (mtu == 250) mtuChanged.complete();
    });
    await mtuChanged.future.timeout(const Duration(seconds: 2)).catchError(
        (dynamic error) =>
            debugPrint(error.toString())); // set timeout and catch exception
    mtuStreamSubscription?.cancel();

closed 2021-09-17 by HninNweAye

Enemias-Itzep, 2022-02-27:
@HninNweAye Where did you make this change?