pauldemarco/flutter_blue #938: Can not run my application

fedorstepanovv, 2021-09-16:
Before I had been using  flutter_blue: 0.8.0 for a while, but then i couldnt run my app because of the issue below:
* What went wrong:
Execution failed for task ':flutter_blue:generateDebugProto'.
> Could not resolve all files for configuration ':flutter_blue:protobufToolsLocator_protoc'.
   > Could not find protoc-3.11.4-osx-aarch_64.exe (com.google.protobuf:protoc:3.11.4).
     Searched in the following locations:
         https://jcenter.bintray.com/com/google/protobuf/protoc/3.11.4/protoc-3.11.4-osx-aarch_64.exe

It seems like there are some sort of problems with jcenter, because the url right from the error doesnt work.

sanatChoubey, 2021-09-16:
@fedorstepanovv  have you got a solution?

fedorstepanovv, 2021-09-16:
No, I don’t. But on my intel mac it works perfectly fine. 

> @fedorstepanovv have you got a solution?
> I have change jcenter to maven but still getting the same issue
> [image]

peekpt, 2021-09-16:
https://github.com/pauldemarco/flutter_blue/pull/899

fedorstepanovv, 2021-09-16:
> #899

@peekpt , where exactly do i need to paste this line? As i unterstood this part of code must be in android/build.gradle. But it doesnt seem like a line that was written in the reply right for build.gradle

peekpt, 2021-09-16:
Fork code and apply PullReq to your fork. Or use the authors fork/branch that includes the PR in pubspec

sanatChoubey, 2021-09-17:
@fedorstepanovv  I found a solution
[image]
[image]

first go to your-> flutter project/android/build.gradle
and add ext.protoc_platform = osx-x86_64
and then open android folder in android studio
and in side bar you will find flutter blue and in flutter blue open build.gradle and add second image code.

fedorstepanovv, 2021-09-26:
The issue was solved, i tried to add lines, which @sanatChoubey has shared, but in my case i opened android project folder in android studio and synchronized my build.gradle file. It took around 14 mins  to install everything. So, after that step, my app launches without any addional lines.

closed 2021-09-26 by fedorstepanovv