pauldemarco/flutter_blue #939: Send text between devices

staminna, 2021-09-16:
Hi

Is is possible to send a string of text between connected devices?

peekpt, 2021-09-16:
Please use discord server for this questions.
Bluetooth is 8 bit you need to convert everything to byte Array Uint8List

HninNweAye, 2021-09-17:
await characteristics.write(utf8.encode('text'));
utf8.encode() will convert your string to ASCII codes (int list).

farhantandia, 2021-12-28:
Hi, I'm really new in flutter, and just working on BLE device, anyone could share simple implementation to send and receive the message between phone and ble?? which service/function do I need to use? this is my current app looks like: Thankyou!
[image]