pauldemarco/flutter_blue #947: Android build fails on Mac M1

EinatK, 2021-09-23:
This issue happened in other packages as well but was fixed by them. 
Please fix ASAP, package is unusable for M1 macs 

Error 

```
Execution failed for task ':flutter_blue:generateDebugProto'.
> Could not resolve all files for configuration ':flutter_blue:protobufToolsLocator_protoc'.
   > Could not find protoc-3.11.4-osx-aarch_64.exe (com.google.protobuf:protoc:3.11.4).
     Searched in the following locations:
         https://jcenter.bintray.com/com/google/protobuf/protoc/3.11.4/protoc-3.11.4-osx-aarch_64.exe
```

Flutter doctor -v 

```
[✓] Flutter (Channel stable, 2.2.3, on macOS 11.5.2 20G95 darwin-arm, locale
    en-US)
    • Flutter version 2.2.3 at /Users/einatkidron/flutter
    • Framework revision f4abaa0735 (3 months ago), 2021-07-01 12:46:11 -0700
    • Engine revision 241c87ad80
    • Dart version 2.13.4

[✓] Android toolchain - develop for Android devices (Android SDK version 30.0.0)
    • Android SDK at /Users/einatkidron/Library/Android/sdk
    • Platform android-30, build-tools 30.0.0
    • ANDROID_HOME = /Users/einatkidron/Library/Android/sdk
    • Java binary at: /Applications/Android
      Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7249189)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 12.5.1, Build version 12E507
    • CocoaPods version 1.10.0

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[✓] Android Studio (version 2020.3)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7249189)

[✓] VS Code (version 1.60.1)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.26.0

[✓] Connected device (2 available)
    • iPhone 11 Pro Max (mobile) • D2869708-366F-4153-AA5C-2995E8D68046 • ios
      • com.apple.CoreSimulator.SimRuntime.iOS-14-5 (simulator)
    • Chrome (web)               • chrome                               •
      web-javascript • Google Chrome 93.0.4577.82

• No issues found!
```

MacBook Pro (13-inch, M1, 2020)
MacOS BigSur 11.5.2

peekpt, 2021-09-23:
https://github.com/pauldemarco/flutter_blue/pull/899

boskokg, 2021-09-26:
https://github.com/pauldemarco/flutter_blue/issues/929#issuecomment-927268881

MrCsabaToth, 2021-12-24:
@jason19970210 build.gradle is part of flutter_blue repo.

borisllona, 2021-12-29:
> @EinatK Actually this error is not came from flutter_blue, you have to change the `build.gradle` file of flutter_blue gradle config
> 
> Step 1. Use Android Studio to open the `android` folder in your target flutter directory Step 2. There is a vertical `Gradle` tab showing on the very right side of Android Studio Panel
> 
> the structure will shown like below :
> 
> ```
> | - android
>     | - app
>     | - flutter_blue                                                                           <--- right click
>         | - Dependencies
>             | - io.flutter:flutter_embedding_debug...
>             | - ...
>             | - ...
>     | - shared_preferences
> ```
> 
> please right click on the `flutter_blue` then choose `Open Gradle Config`
> 
> Step 3. There are 2 lines you need to modify in your `build.gradle` at the bottom of the file :
> 
> ```
> protobuf {
>     protoc {
>         // artifact = 'com.google.protobuf:protoc:3.11.4'
>         artifact = 'com.google.protobuf:protoc:3.17.3'
>     }
>     generateProtoTasks {
>         all().each { task ->
>             task.builtins {
>                 java {
>                     option "lite"
>                 }
>             }
>         }
>     }
> }
> 
> dependencies {
>     // implementation 'com.google.protobuf:protobuf-javalite:3.11.4'
>     implementation 'com.google.protobuf:protobuf-javalite:3.17.3'
> }
> ```
> 
> First change : `'com.google.protobuf:protoc:3.11.4'` to `'com.google.protobuf:protoc:3.17.3'` Second change : `'com.google.protobuf:protobuf-javalite:3.11.4'` to `'com.google.protobuf:protobuf-javalite:3.17.3'`
> 
> Hope this help !!
> 
> origin post : [protocolbuffers/protobuf#8062 (comment)](https://github.com/protocolbuffers/protobuf/issues/8062#issuecomment-892689597)

I can't see flutter_blue in my android folder, I have it in pubsecyml. How can we fix this? Because I have an M1 Mac and still not working

sahilnagpal17, 2022-01-03:
@borisllona Can you check in the External libraries in Android Studio?

```
External Libraries > Flutter Plugins > flutter_blue-0.8.0 > android > build.gradle
```

borisllona, 2022-01-04:
I'm sorry but this path does not exist :(. I am using the last version in a Mac M1.

Hytro, 2022-01-28:
> I'm sorry but this path does not exist :(. I am using the last version in a Mac M1.

Make sure you only open the Android folder of the Flutter app in Android Studio to get the Gradle tab on the right, not the whole flutter project.

azamatdev, 2022-02-15:
@sahilnagpal17 and @Hytro thank you for clarifying. I was confused which gradle file needs to be changed at first. I changed the external libraries gradle file of the flutter_blue package. Everything worked)

borisllona, 2022-02-16:
Yes, worked! Thanks for the clarification

MaxiStarling56, 2022-03-23:
> implementation 'com.google.protobuf:protobuf-javalite:3.17.3'

Worked, Thanks a lot!

paakjis, 2022-04-27:
I have flutter_blue: 0.7.1+1. This is what I did to get it working on M1.
1. File -> Open. Open the 'android' folder in Android Studio. You will get the Gradle button on top right that everyone is talking about here.
2. There you do the steps that are mentioned here https://github.com/pauldemarco/flutter_blue/issues/947#issuecomment-948480585
3. Because I used an old version of flutter_blue, I did not have to change only 2 lines, I replaced all the old 

```
protobuf {
    // Configure the protoc executable
    protoc {
        // Download from repositories
        artifact = 'com.google.protobuf:protoc:3.9.1'
    }
    plugins {
        javalite {
            // The codegen for lite comes as a separate artifact
            artifact = 'com.google.protobuf:protoc-gen-javalite:3.0.0'
        }
    }
    generateProtoTasks {
        all().each { task ->
            task.plugins {
                javalite { }
            }
        }
    }
}

dependencies {
    implementation 'com.google.protobuf:protobuf-lite:3.0.1'
    implementation 'androidx.appcompat:appcompat:1.0.0'
    api 'androidx.core:core:1.0.1'
}
```

To the new lines, that were suggested previously. And it all works.
```
protobuf {
    protoc {
        artifact = 'com.google.protobuf:protoc:3.17.3'
    }
    generateProtoTasks {
        all().each { task ->
            task.builtins {
                java {
                    option "lite"
                }
            }
        }
    }
}

dependencies {
    implementation 'com.google.protobuf:protobuf-javalite:3.17.3'
}

```

soubinkittiphanh, 2022-05-05:
Where can i find build.gradle of flutter_blue in vscode?

boskokg, 2022-05-05:
@soubinkittiphanh use flutter_blue_plus plugin

TamirlanBir, 2022-08-04:
> @EinatK Actually this error is not came from flutter_blue, you have to change the `build.gradle` file of flutter_blue gradle config
>
> Step 1. Use Android Studio to open the `android` folder in your [...]

After "flutter pub get" - my dile .packages and Dart Packages(Flutter plugins )  from External Libraries disappeared.
How can I fix it or where is this files ?