pauldemarco/flutter_blue #951: Fork List (with enhancements and PR)

peekpt, 2021-09-25:
Since this plugin's development is on hold, please post here what are the Fork you are using and reason.

 -- fork:
 
 -- reason?:

peekpt, 2021-09-25:
fork: https://github.com/MZachmann/flutter_blue

reason: It has updated to last PR, can control Bluetooth and can read RSSI from peripheral

boskokg, 2021-09-26:
please read this: https://github.com/pauldemarco/flutter_blue/issues/929#issuecomment-927268881

boskokg, 2021-09-26:
@peekpt can you please close this issue?

peekpt, 2021-09-26:
@boskokg I reopen the issue because I found  MZachmann's fork differences:
 - mtuRequest returns the MTU 
 - turnOn and turnOff Bluetooth on Android

boskokg, 2021-09-26:
I will add them today. Please continue writing there, not here so others can catch the entire discussion in one place.

peekpt, 2021-09-26:
@boskokg  Ok thank you for your job. I will close this

closed 2021-09-26 by peekpt