pauldemarco/flutter_blue #954: Reading CBOR data is not possible

samigimix, 2021-09-28:
Hello

I have a problem with reading CBOR data. The "read" function is based on a list of integers but it doesn't read bytes and it converts automatically to int and then stores it. Is there any solution to read CBOR data (list of bytes) through Bluetooth?

Thanks

peekpt, 2021-09-28:
Has you should know bluetooth communicates with 8 bit. If you want to send or receive, you need to convert to Uint8List (byte array) which isn't a growable list, it's immutable however, you convert using Uint8List.fromList(). There's no type of Hex  in dart, so you will be working with int. But it's possible to assign int a= 0xA; check on pub.dev the Hex.decode lib , as well the converters toRadix() String.codebits

samigimix, 2021-10-05:
The issue was with the sender device, but it sends multiple log packages, and the read function only reads the last package, not the previous ones. I tried notifying function, larger MTU size too but didn't work.

Here is my code:

> var value = await c.read();
> final cbor = Cbor();
> final payloadBuffer = Uint8Buffer();
> payloadBuffer.addAll(value);
> cbor.decodeFromBuffer(payloadBuffer);
> String str = cbor.decodedToJSON();
> var js = jsonDecode(str);