pauldemarco/flutter_blue #965: not showing device list on android device

RRBavda, 2021-10-18:
Hello there,

I am trying to search for devices on my android phone. There is no list showing on my android phone. But, when I run the same code on my iOS phone, it shows the device in the list.

May I know why is it happening?

Note: My Android OS version is 10.

peekpt, 2021-10-18:
Did you edit androidmanifest

RRBavda, 2021-10-20:
main.dart

`
import 'dart:async';
import 'dart:math';

import 'package:flutter/cupertino.dart';
import 'package:flutter/material.dart';
import 'package:flutter_blue/flutter_blue.dart';
import 'package:notification_permissions/notification_permissions.dart';
import 'package:test_ble/notificationApi.dart';

import 'package:test_ble/widgets.dart';

import 'package:flutter_local_notifications/flutter_local_notifications.dart';
// import 'package:flutter/cupertino.dart';

// late FlutterLocalNotificationsPlugin flutterLocalNotificationsPlugin;

void main() {
  //flutterLocalNotificationsPlugin = FlutterLocalNotificationsPlugin();
  runApp(MyApp());
}

class MyApp extends StatelessWidget {
  @override
  Widget build(BuildContext context) {
    return MaterialApp(
      debugShowCheckedModeBanner: false,
      color: Colors.lightBlue,
      home: StreamBuilder<BluetoothState>(
          stream: FlutterBlue.instance.state,
          initialData: BluetoothState.unknown,
          builder: (c, snapshot) {
            final state = snapshot.data;
            if (state == BluetoothState.on) {
              return FindDevicesScreen();
            }
            return BluetoothOffScreen(state: state);
          }),
    );
  }
}

class BluetoothOffScreen extends StatelessWidget {
  const BluetoothOffScreen({Key? key, this.state}) : super(key: key);

  final BluetoothState? state;

  @override
  Widget build(BuildContext context) {
    return Scaffold(
      backgroundColor: Colors.lightBlue,
      body: Center(
        child: Column(
          mainAxisSize: MainAxisSize.min,
          children: <Widget>[
            Icon(
              Icons.bluetooth_disabled,
              size: 200.0,
              color: Colors.white54,
            ),
            Text(
              'Bluetooth Adapter is ${state != null ? state.toString().substring(15) : 'not available'}.',
              style: Theme.of(context)
                  .primaryTextTheme
                  .subhead
                  ?.copyWith(color: Colors.white),
            ),
          ],
[... 685 lines cut ...]

class AdapterStateTile extends StatelessWidget {
  const AdapterStateTile({Key? key, required this.state}) : super(key: key);

  final BluetoothState state;

  @override
  Widget build(BuildContext context) {
    return Container(
      color: Colors.redAccent,
      child: ListTile(
        title: Text(
          'Bluetooth adapter is ${state.toString().substring(15)}',
          style: Theme.of(context).primaryTextTheme.subhead,
        ),
        trailing: Icon(
          Icons.error,
          color: Theme.of(context).primaryTextTheme.subhead?.color,
        ),
      ),
    );
  }
}

`

dinhvuvan1996, 2021-10-21:
I have same issue with RRBavda. Please help me to check.  
https://github.com/dinhvuvan1996/Dieu_Khien_Led_Be_Ca_Thuy_Sinh.git

halloweens1, 2021-10-21:
@RRBavda have you solved it yet? cuz I have the same problem, so confusing, it should not work this way, it's not working on my android device with Android 11 (means cannot discover any other bluetooth devices by scanning), but it works pretty well on my other android device with Android 7, weird

halloweens1, 2021-10-22:
@RRBavda bro, I solved it, by chance I clicked into one of those toturials online, I noticed one thing that its minSDKVersion changed, so I just tried to change it as the tutorail said, and then, it works, and by the way, actually the author has told us on the readme file,  `Getting Started
Change the minSdkVersion for Android
Flutter_blue is compatible only from version 19 of Android SDK so you should change this in android/app/build.gradle:

Android {
  defaultConfig {
     minSdkVersion: 19`, I changed it to version 28 then works. This is a bad habit that not to read the readme file throughly, especially the setup/getting started part, won't happend again, next time will be more careful.

halloweens1, 2021-10-23:
@RRBavda Now I seem to find my own problem, the reason my dev app(using flutter_blue) that cannot discover any other  bluetooth devices by scanning is the game mode on my phone(vivo device Android 11), so weird that my dev app is not designed for game, but it's added to the game mode box, when I kicked it out of the game mode box, issue's gone, works pretty well, I had noticed, but I think that won't affect the app, the answer is I was wrong, hope this may give you some help

deucks, 2021-11-01:
for me when i open bluetooth settings and conduct a search from there, my app in the background is able to receive new devices. Once i close settings, new device discovery stops in the app. Any clues to why this may be?

Jeeva111, 2021-11-24:
Turn ON GPS it's working 
https://github.com/pauldemarco/flutter_blue/issues/908

RRBavda, 2021-11-24:
@Jeeva111 
Thank you for your help but, I've already tried that option.

Techmaniac154, 2021-12-14:
> Turn ON GPS it's working #908

this worked for me thanks