pauldemarco/flutter_blue #977: Subscribe to characteristic notifications. Receiving incomplete data.

pavluchenkor, 2021-11-04:
characteristics[characteristicIndex(configCharacteristicUUID)].value.listen((value) {print(ascii.decode(value).split('/'););}
and 
Future<List<int>> value = characteristics[characteristicIndex(configCharacteristicUUID)].read();
print(ascii.decode(value).split('/');

Give me different results, data vary in length. At the same time, reading and receiving notifications in the BLE Device Info program gives me the same data.

On the server side, everything is implemented in one function, the data goes from it to the characteristic for reading and the characteristic for notifications.

Everything seems very simple, but for some reason it is not.
I used a third party program to validate my data and made sure I was sending the correct information to both characteristics, but I get different data in my program.
Help!
What else can you check?
[image]
Read----------------
[image]
Notify----------------
[image]

radianceltd, 2021-11-16:
Hello, why did I report this error when I connected to Bluetooth.
```
[ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: FormatException: The format is invalid
#0      Guid._fromString (package:flutter_blue/src/guid.dart:45:7)
#1      new Guid (package:flutter_blue/src/guid.dart:15:39)
#2      new BluetoothCharacteristic.fromProto (package:flutter_blue/src/bluetooth_characteristic.dart:35:27)
#3      new BluetoothService.fromProto.<anonymous closure> (package:flutter_blue/src/bluetooth_service.dart:19:29)
#4      MappedListIterable.elementAt (dart:_internal/iterable.dart:413:31)
#5      ListIterator.moveNext (dart:_internal/iterable.dart:342:26)
#6      new _GrowableList._ofEfficientLengthIterable (dart:core-patch/growable_array.dart:188:27)
#7      new _GrowableList.of (dart:core-patch/growable_array.dart:150:28)
#8      new List.of (dart:core-patch/array_patch.dart:50:28)
#9      ListIterable.toList (dart:_internal/iterable.dart:213:44)
#10     new BluetoothService.fromProto (package:flutter_blue/src/bluetooth_service.dart:20:14)
#11     BluetoothDevice.discoverServices.<anonymous closure>.<anonymous closure> (package:flutter_blue/src/bluetooth_device.dart:67:38)
#12     MappedListIterable.elementAt (dart:_internal/iterable.dart:413:31)
#13     ListIterator.moveNext (dart:_internal/iterable.dart:342:26)
#14     new _GrowableList._ofEfficientLengthIterable (dart:core-patch/growable_array.dart:188:27)
#15     new _GrowableList.of (dart:core-patch/growable_array.dart:150:28)
#16     new List.of (dart:core-patch/array_patch.dart:50:28)
#17     ListIterable.toList (dart:_internal/iterable.dart:213:44)
#18     BluetoothDevice.discoverServices.<anonymous closure> (package:flutter_blue/src/bluetooth_device.dart:67:69)
#19     _MapStream._handleData (dart:async/stream_pipe.dart:213:31)
#20     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#21     _rootRunUnary (dart:async/zone.dart:1436:47)
#22     _CustomZone.runUnary (dart:async/zone.dart:1335:19)
#23     _CustomZone.runUnaryGuarded (dart:async/zone.dart:1244:7)
#24     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:341:11)
#25     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#26     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#27     _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
#28     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#29     _rootRunUnary (dart:async/zone.dart:1436:47)
#30     _CustomZone.runUnary (dart:async/zone.dart:1335:19)
#31     _CustomZone.runUnaryGuarded (dart:async/zone.dart:1244:7)
#32     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:341:11)
#33     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#34     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#35     _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
#36     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#37     _rootRunUnary (dart:async/zone.dart:1436:47)
#38     _CustomZone.runUnary (dart:async/zone.dart:1335:19)
#39     _CustomZone.runUnaryGuarded (dart:async/zone.dart:1244:7)
#40     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:341:11)
#41     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#42     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#43     _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
#44     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#45     _rootRunUnary (dart:async/zone.dart:1436:47)
#46     _CustomZone.runUnary (dart:async/zone.dart:1335:19)
#47     _CustomZone.runUnaryGuarded (dart:async/zone.dart:1244:7)
#48     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:341:11)
#49     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#50     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#51     _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
#52     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#53     _rootRunUnary (dart:async/zone.dart:1436:47)
#54     _CustomZone.runUnary (dart:async/zone.dart:1335:19)
#55     _CustomZone.runUnaryGuarded (dart:async/zone.dart:1244:7)
#56     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:341:11)
#57     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#58     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#59     _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
#60     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#61     _rootRunUnary (dart:async/zone.dart:1436:47)
#62     _CustomZone.runUnary (dart:async/zone.dart:1335:19)
#63     _CustomZone.runUnaryGuarded (dart:async/zone.dart:1244:7)
#64     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:341:11)
#65     _DelayedData.perform (dart:async/stream_impl.dart:591:14)
#66     _StreamImplEvents.handleNext (dart:async/stream_impl.dart:706:11)
#67     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:663:7)
#68     _rootRun (dart:async/zone.dart:1420:47)
#69     _CustomZone.run (dart:async/zone.dart:1328:19)
#70     _CustomZone.runGuarded (dart:async/zone.dart:1236:7)
#71     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1276:23)
#72     _rootRun (dart:async/zone.dart:1428:13)
#73     _CustomZone.run (dart:async/zone.dart:1328:19)
#74     _CustomZone.runGuarded (dart:async/zone.dart:1236:7)
#75     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1276:23)
#76     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
#77     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
```
Can you help me?  thank you.

radianceltd, 2021-11-16:
Error occurred at:
```
 var services = await device.discoverServices();
```

peekpt, 2021-11-16:
What's ascii.decode?

pravin-sorigin, 2022-02-21:
Try using `device.requestMtu()`