pauldemarco/flutter_blue #978: Write characteristic failed

iameerhamza, 2021-11-06:
i am using flutter blue with esp 32 and it send messages successfully, the problem arises when i start notifying. the data coming from esp32 is continuously so when i try to send data it show the error.

Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null, null)
E/flutter ( 1092): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:597:7)
E/flutter ( 1092): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:158:18)
E/flutter ( 1092): <asynchronous suspension>
E/flutter ( 1092): #2      BluetoothCharacteristic.write (package:flutter_blue/src/bluetooth_characteristic.dart:122:18)
E/flutter ( 1092): <asynchronous suspension>
E/flutter ( 1092): 

any suggestions are welcome.

peekpt, 2021-11-06:
Are you using the same characteristic to notify and write? The recommended thing to do is having one to write (tx) and other to read notify (rx)

iameerhamza, 2021-11-06:
yes, i have different characteristics for both rx and tx, 
as i mentioned characteristics.write works perfectly until or unless i set characteristics.notify to "true"

peekpt, 2021-11-06:
You mean setting rx characteristic notify to true? not the tx. 
To enable notify indicate on rx your device must have a 2902 descriptor on the rx char.

Use the discord channel for questions. 
use the updated fork https://github.com/pauldemarco/flutter_blue/issues/929#issuecomment-930759129

closed 2021-11-06 by iameerhamza