pauldemarco/flutter_blue #984: writeWithoutResponse: true error

mighty-zh, 2021-11-19:
CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: true, write: true, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}
this is print log of characteristic, but when set writeWithoutResponse: true got this log PlatformException(write_characteristic_error, writeCharacteristic failed, null, null) and then device is disconnected; how can I write data withoutresponse?

peekpt, 2021-11-19:
Check your device with another bluetooth app if the behaviour is the same.

mighty-zh, 2021-11-22:
I have tested more than one device, they all works fine with WriteWithOutResponse on native iOS and Android bluetooth app, but not work on flutter app and get the same error like I said before.     Also, the version of  flutter_blue I use is 0.8.0

mighty-zh, 2021-11-22:
> Check your device with another bluetooth app if the behaviour is the same.

I have tested more than one device, they all works fine with WriteWithOutResponse on native iOS and Android bluetooth app, but not work on flutter app and get the same error like I said before. Also, the version of flutter_blue I use is 0.8.0

mighty-zh, 2021-11-22:
> Check your device with another bluetooth app if the behaviour is the same.

 I have tested more than one device, they all works fine with WriteWithOutResponse on native iOS and Android bluetooth app, flutter app on iOS works, but not work on Android and get the same error like I said before. Also, the version of flutter_blue I use is 0.8.0

mighty-zh, 2021-11-24:
I have solved this problem, I have add a condition in BluetoothCharacteristic Class at line 123, only return result on iOS,  and the code like this
if (Platform.isIOS && type == CharacteristicWriteType.withoutResponse) {
      return result;
}
when return result on android may cause the package lose, and that makes the problem.
so on android I get WriteCharacteristicResponse and send next package.
@peekpt

closed 2021-11-24 by mighty-zh