pauldemarco/flutter_blue #986: BLE didn`t work in release version android

TeslaWorks239, 2021-11-24:
BLE do nothing in release but in debug version everything fine. I use 0.8.0 version of Flutter blue

XML:
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.lamply.lamply">
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
   <application

build.gradle
  buildTypes {
        release {
            // TODO: Add your own signing config for the release build.
            // Signing with the debug keys for now, so `flutter run --release` works.
            signingConfig signingConfigs.debug
            shrinkResources false
            minifyEnabled false
        }
    }
}

qinjinze, 2021-11-28:
The Bluetooth status can be obtained by debugging and installing on the real machine, but the Bluetooth status cannot be obtained after compiling the apk

peekpt, 2021-11-28:
You need to setup the proguard rules, search the issues there's something about it

TeslaWorks239, 2021-11-28:
Thank you! I add this and everything work fine
[image]

closed 2021-11-28 by TeslaWorks239

FarhanRiaaz, 2021-11-29:
Why don't you put 
`signingConfig signingConfigs.release`

instead of this
`signingConfig signingConfigs.debug`

littleantsos, 2022-03-25:
> Thank you! I add this and everything work fine [image]
It worked

FredyKonig, 2022-07-05:
Do we also have a fix for the IOS build?

boskokg, 2022-07-05:
use flutter_blue_plus

FredyKonig, 2022-07-05:
> use flutter_blue_plus

Thanks, we tried using flutter_blue_plus but it doesn't seem to have fixed it in the IOS release version.