pauldemarco/flutter_blue #989: device names is empty ? WHHYYYYY

muktharrox, 2021-11-26:
Flutter blue, doesn't show device names neither in device.name nor in advertisementData.localName WHY ?
I'm trying to read an OBD II module and I'm having a hard time doing that, does anyone have an example about this?

andrewzakhartchouk, 2021-11-26:
Can you provide some code of what you have tried to do exactly?

f1forhelp, 2021-11-30:
Same issue with me, with Bluetooth serial its showing name but by using this package only some ID's are displaying.

f1forhelp, 2021-11-30:
> Can you provide some code of what you have tried to do exactly?

You can check with the same example code.

peekpt, 2021-11-30:
This is only for BLE devices only

pdhash, 2022-01-11:
Same issue with me, i can not get it.

ronaldmaymone, 2022-01-21:
@muktharrox I'm trying to do the exact same thing as you. Can you say if you managed to do that? Thx!

yukitaka13-1110, 2022-05-11:
I'm having same issue. Any update?

mandreshope, 2022-08-12:
Same issue on flutter_blue 0.8.0