pauldemarco/flutter_blue #990: Bluetooth Characteristic  setNotifyValue only showing 21 first characters?

MarioNepomuceno, 2021-11-26:
It returns me

`{"PumpType":"Demopri`

While I should get

`{"PumpType":"Demopring","Power":1,"TempWater":28.5,"TempAmbient":22.0,"TempSet":29.0,"ErrorMessages":""}`

peekpt, 2021-11-27:
Change mtu to be like 500 or send your data in chunks

MarioNepomuceno, 2021-11-29:
```
for (BluetoothService service in services) {
 for (BluetoothCharacteristic characteristic in service.characteristics) {
          if (characteristic.properties.notify) {
            cNotify = characteristic;

            sub = cNotify.value.listen( (value) {
                  **I don't know how many times the debug passes within this context to know when I've just received the data**
              },
            );
            await cNotify.setNotifyValue(true);

            setState(() {});
       }
  }
}
```

Enemias-Itzep, 2022-02-27:
> Change mtu to be like 500 or send your data in chunks

> Change mtu to be like 500 or send your data in chunks

How do I change the MTU?

Enemias-Itzep, 2022-02-27:
@peekpt How do i change the MTU?

peekpt, 2022-02-27:
> @peekpt How do i change the MTU?

Seriously?
Look at read me