pauldemarco/flutter_blue #996: How to Get/receive data from Bluetooth device?

Logamuhilan, 2021-12-11:
I am using flutter_blue to connect and get Bluetooth device data, my problem is i don't know **the method to get data** from it. I have used "await c.read" it gives only device name, and "await d.read" it gives only empty array. here "c" and "d" is characteristics. I think read is not a method i think. Is there any method to get data from my device? Cheers in advance ; )

oooldCaption, 2021-12-16:
like this : 
```
//  characteristic --> your's notification characteristic
await characteristic.setNotifyValue(true);
characteristic.value.listen((value) {
// this   value   are the data uploaded by the Bluetooth device 
    // do something with new value
});
```

Logamuhilan, 2021-12-24:
> like this : 
> ```
> //  characteristic --> your's notification characteristic
> await characteristic.setNotifyValue(true);
> characteristic.value.listen((value) {
> // this   value   are the data uploaded by the Bluetooth device 
>     // do something with new value
> });
> ```

Thanks for your response 👍🏼

closed 2022-04-16 by Logamuhilan

dinhvuvan1996, 2022-06-11:
Hi @Logamuhilan, the same problem is occurring to me. I tried to use @oooldCaption 's solution but it not work, I can't pass the line: await characteristic.setNotifyValue(true);. Please help me

oooldCaption, 2022-09-09:
code like this: 
do u add `async ` on you function ?

  getNotify() async{
    await characteristic.setNotifyValue(true);
    characteristic.value.listen((value) {
// this   value   are the data uploaded by the Bluetooth device 
      // do something with new value
    });
  }